mcguirepr89/BirdNET-Pi #1030: Unable to send whatsapp notifications as apprise is an old version

wiltonp80, 2023-09-21:
Unable to undertake several notifications in the apprise suite in particular for me, a whatsapp message.
This is probably a versioning of apprise included which is v1.2.1

When it is placed in Settings -> Apprise Notifications Configuration, it causes BirdNET-Pi to crash as soon as it attempts the first notification.
Investigating based on the appraise troubleshooting it was possible to manually test appraise commands and discover the issue.

**To Reproduce**
1) run '/home/pi/BirdNET-Pi/birdnet/bin/apprise -V'
2) run '/home/pi/BirdNET-Pi/birdnet/bin/apprise -b "Test Message" "whatsapp://<token>/<phone id>/44<recipient>" '
3) run '/home/pi/BirdNET-Pi/birdnet/bin/apprise -l'

** Output **
1) Apprise v1.2.1
2) ERROR - Unparseable URL whatsapp://<token>@<phone id>/44<recipient>
3) Does not include whatsapp

**Expected behavior**
1) Apprise v1.5.0
2) No error our output but it should send whatapp message
3) Expect whatsapp in the available services

**Your build**
Raspberry Pi 4 B (PCB 1.5 Sony UK)
8GB

Can a later release be incorporated into a forthcoming update please so

AugustT, 2023-12-27:
I'm having the same problem. I tried to upgrade the version of apprise using the web terminal, but no success.

AugustT, 2023-12-27:
[image]

So using the web console it says that the version is updated, but am I right in thinking that this has not updated the package in the right place?

Der-Sven, 2024-03-03:
You are right.
Birdnet installs the packages into a virtual envirnoment. any other installs are not used.
```
cd ~/BirdNET-Pi
python3 -m venv birdnet
source ./birdnet/bin/activate
pip3 install -U apprise==1.6.0
```

Pete-the-Geek, 2024-03-03:
Worked, like a charm. 

Thanks @Der-Sven