mcguirepr89/BirdNET-Pi #1033: Unable to save Apprise notification body configuration

rappazzo, 2023-09-23:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
I am able to save the notification body for my apprise messages.   I put this into the field, and I am able to successfully send (and receive) a test message.  However, when I click the "Update Settings" button, once the response comes back, the notification body is cleared.

Here is the content of the body:
```
{"scientificName":"$sciname","commonName":"$comname","confidenceScore":"$confidence","link":"$listenurl","date":"$date","time":"$time","week":"$week","latitude":"$latitude","longitude":"$longitude","image":"$flickrimage"}
```
I did have a configuration successfully saved previously, but I tried to modify it today, and it is now exhibiting this behavior.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Tools.
2. Click on Settings
3. Scroll down to Notifications, and fill out "Apprise Notification Configuration" and the "Notification Body"
4. Click "Send Test Notification"
5. Verify that the message was sent and received
6. Scroll down to the bottom
7. Click "Update Settings"
8. When the save completes, scroll up and observe that the "Notification Body" is now blank.

**Expected behavior**
The Notification Body should save and be retained in the field

**Additional context**
On the receiver side, I did notice that the test message that I received appeared to be escaped json (`\{"scientificName":"Anser caerulescens", ... \}`).  To make sure that this wasn't a receiver side issue, I ran a test from the command line (ssh'd in) 
```
$HOME/BirdNET-Pi/birdnet/bin/apprise -vv -b "${PAYLOAD}" "mqtt://${HOMEASSISTANT}/birdnet/sightings"
```
This was received without escaped `{` and `}` characters

**Your build**
Raspberry Pi 3 B+ (PCB 1.3 Sony UK)
Running version: [762732c13434c5ab81d667da952e2345a04519cb](https://github.com/mcguirepr89/BirdNET-Pi/commit/762732c13434c5ab81d667da952e2345a04519cb)

rappazzo, 2023-09-23:
Is there a way that I can workaround the ui configuration by adjusting the setting in a file or database somewhere?

johanneseckert, 2023-10-11:
I had the same issue. I think it's not correctly saving/displaying the value because of some some special characters in the JSON format. I also never got a notification (sending to MQTT as the target)

I fixed this by simply using a comma separated list as the body. I'm parsing the notification in Nodered and changed it to CSV format there.

However, $flickrimage is empty for me, but the rest works — see  https://github.com/mcguirepr89/BirdNET-Pi/issues/1042

barrywoolgar, 2023-10-28:
Thanks @rappazzo, I edited my `birdnet.conf` to fix the unnecessary(?) encoding of double quotes and it's working:

Simplified example:
```
APPRISE_NOTIFICATION_BODY='{"Common_Name": "$comname"}'
```

I then used "Tools -> Services -> Restart Core Services" and it started working as expected. 

There's a good chance this will break again next time the configuration file is rewritten, so I'm going to avoid changing any settings for a while. If that gets annoying I'll go the NodeRed route (or, you know, try to make a bugfix PR here)