mcguirepr89/BirdNET-Pi #1036: Losing Microphone Input After Crash / Reboots

bonehead1996, 2023-10-03:
**Describe the bug**
When rebooting the Pi (using the web interface reboot button and/or a hardware power switch) BirdNET will occasionally come back up but be detecting no sound input leading to essentially blank spectrograms with some interference present. No detections are made either owing to the lack of input. 

Restarting the Pi, or starting and stopping individual services does not remedy this issue, and neither does unplugging and re-plugging the USB microphone. The only thing that fixes the issue is re-flashing the SD card with a new image and starting from fresh.

**To Reproduce**
I can't reliably reproduce this but it has happened probably 6 times in about 30 reboots of the Pi, which is unhelpful as I'm trying to run a portable field-deployable instance of BirdNET. 

**Expected behavior**
Safely rebooting the Pi shouldn't cause loss of microphone / sound input

**Screenshots**
[image]

**Your build**
RPi Zero 2 W with "MOVO" generic USB microphone.

**Code or log snippets**
```
Attempting to restart the live audio stream produces the following:

```

ehpersonal38, 2023-10-09:
What's actually happening, I think, is that the gain of the microphone is being turned down to a very low level. You can see in the spectogram there actually is some data there. The recommended fix is to turn the gain back up - https://github.com/mcguirepr89/BirdNET-Pi/wiki/Adjusting-your-sound-card.

bonehead1996, 2023-10-10:
> What's actually happening, I think, is that the gain of the microphone is being turned down to a very low level. You can see in the spectogram there actually is some data there. The recommended fix is to turn the gain back up - https://github.com/mcguirepr89/BirdNET-Pi/wiki/Adjusting-your-sound-card.
> 
> As to what causes it, I'm not exactly sure. Though I will say that the exact same issue was occuring to me back when I used a MOVO M1, so it may be limited to this brand, or cheaper mics.

You're 100% correct, thank you!

I don't know why I didn't twig that could be the issue. Interesting you had the same issue with a MOVO mic. I'll have to see if I can track down an alternative and test to see if the same issue occurs.