mcguirepr89/BirdNET-Pi #1038: The installation exited unsuccessfully.

kinteki, 2023-10-04:
### Discussed in https://github.com/mcguirepr89/BirdNET-Pi/discussions/1037

<div type='discussions-op-text'>

<sup>Originally posted by **kinteki** October  3, 2023</sup>
Hi, there,

New installation exited. Brand new Raspbian OS 64 bit - lite with 128 GB SSD. 
Please help! 

zoli@zoli:~ $ curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/m                                                                                                    ain/newinstaller.sh | bash
Cloning into '/home/zoli/BirdNET-Pi'...
remote: Enumerating objects: 132, done.
remote: Counting objects: 100% (132/132), done.
remote: Compressing objects: 100% (123/123), done.
remote: Total 132 (delta 9), reused 64 (delta 5), pack-reused 0
Receiving objects: 100% (132/132), 146.11 MiB | 5.67 MiB/s, done.
Resolving deltas: 100% (9/9), done.
Updating files: 100% (121/121), done.
+ exec
+++ date +%F
++ tee -i installation-2023-10-03.txt
+ set -e
+ my_dir=/home/zoli/BirdNET-Pi
+ export my_dir=/home/zoli/BirdNET-Pi
+ my_dir=/home/zoli/BirdNET-Pi
+ cd /home/zoli/BirdNET-Pi/scripts
++ uname -m
+ '[' aarch64 '!=' aarch64 ']'
+ ./install_config.sh
+ set -e
+ trap 'exit 1' SIGINT SIGHUP
+ echo 'Beginning ./install_config.sh'
Beginning ./install_config.sh
+ birdnet_conf=/home/zoli/BirdNET-Pi/birdnet.conf
++ curl -s4 http://ip-api.com/json
+ json=
+ exit 1
The installation exited unsuccessfully.

zoli@zoli:~ $ bash ~/BirdNET-Pi/scripts/print_diagnostic_info.sh
========== caddy status ==========
Unit caddy.service could not be found.

========== birdnet_analysis status ==========
Unit birdnet_analysis.service could not be found.

========== birdnet_log status ==========
Unit birdnet_log.service could not be found.

========== birdnet_recording status ==========
Unit birdnet_recording.service could not be found.

========== birdnet_server status ==========
Unit birdnet_server.service could not be found.

========== birdnet_stats status ==========
Unit birdnet_stats.service could not be found.

========== chart_viewer status ==========
Unit chart_viewer.service could not be found.

========== extraction status ==========
Unit extraction.service could not be found.

========== web_terminal status ==========
Unit web_terminal.service could not be found.

========== spectrogram_viewer status ==========
Unit spectrogram_viewer.service could not be found.
[... 161 lines cut ...]
Revision        : c03115
Serial          : 1000000027999f56
Model           : Raspberry Pi 4 Model B Rev 1.5

========== System Temperature ==========
/home/zoli/BirdNET-Pi/scripts/print_diagnostic_info.sh: line 39: bc: command not found
CPU Temperature: 46.25°C / °F

========== Extra Info ==========
sudo: /usr/local/bin/extra_info.sh: command not found

========== Connected Microphone Devices ==========
**** List of CAPTURE Hardware Devices ****

null
    Discard all samples (playback) or generate zero samples (capture)

========= Date and Time ==========
Tue  3 Oct 14:10:01 CEST 2023
===========================================

zoli@zoli:~ $ cat /proc/version
Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023

</div>

tastyfishes, 2023-10-09:
Hi, I'm having the exact same issue with the install when trying to install on new Raspberry Pi 4.

kinteki, 2023-10-09:
[image]

http://ip-api.com  is blocked via cable internet.

ehpersonal38, 2023-10-09:
http://ip-api.com/json seems okay for me. The screenshot above is from the https:// version of the site. (modern phone browsers annoyingly redirect most http:// requests to https:// these days, so even if you explicitly put in http:// it's probably still taking you to the https:// site. This isn't a problem in the BirdNET-Pi config as we're using curl on the http:// address).

tastyfishes, 2023-10-10:
I'm fairly certain it wasn't working via http yesterday but glad that it is now. I managed to bypass it and get BirdNET Pi up and running successfully.