mcguirepr89/BirdNET-Pi #1048: The installation exited unsuccessfully -  ERROR: tflite_runtime-2.6.0

s4en, 2023-10-12:
Steps to reproduce the behavior:
1. Use Rapsberry Imager and install Rasperry Pi OS Lite 64
2. Use Putty to connect with the BirdPi (Raspberry Pi4) in SSH Mode
3. Use the Command `curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash` to install
4. After a while i got this Error and the Installation stops:

```
 + pip3 install -U -r /home/bird/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.

```

joostdjong, 2023-10-13:
You could try the Pi OS (**Legacy**, 64-bit) Lite version of the **Bullseye** port. It solved the issue for me.

s4en, 2023-10-14:
> You could try the Pi OS (**Legacy**, 64-bit) Lite version of the **Bullseye** port. It solved the issue for me.

That did the trick.

closed 2023-10-14 by s4en

breimann, 2023-10-21:
I'm having the same issue, but with the Legacy lite version of Bullseye port...

Help??!!!