mcguirepr89/BirdNET-Pi #1049: The installation exited unsuccessfully - tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported

andrebruton, 2023-10-12:
Using a Raspberri Pi 4 4Gb with 64Gb SD Card

Followed the installation exactly. System runs the installation for a long time and then ends with the following...

Establishing a python virtual environment
+ python3 -m venv birdnet
+ source ./birdnet/bin/activate
++ deactivate nondestructive
++ '[' -n '' ']'
++ '[' -n '' ']'
++ '[' -n /usr/bin/bash -o -n '' ']'
++ hash -r
++ '[' -n '' ']'
++ unset VIRTUAL_ENV
++ unset VIRTUAL_ENV_PROMPT
++ '[' '!' nondestructive = nondestructive ']'
++ VIRTUAL_ENV=/home/rpibird/BirdNET-Pi/birdnet
++ export VIRTUAL_ENV
++ _OLD_VIRTUAL_PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
++ PATH=/home/rpibird/BirdNET-Pi/birdnet/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
++ export PATH
++ '[' -n '' ']'
++ '[' -z '' ']'
++ _OLD_VIRTUAL_PS1=
++ PS1='(birdnet) '
++ export PS1
++ VIRTUAL_ENV_PROMPT='(birdnet) '
++ export VIRTUAL_ENV_PROMPT
++ '[' -n /usr/bin/bash -o -n '' ']'
++ hash -r
+ pip3 install -U -r /home/rpibird/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.
rpibird@rpibird01:~ $

I tried running the pip3 install line again, but no luck...

joostdjong, 2023-10-13:
You could try the Pi OS (**Legacy**, 64-bit) Lite version of the **Bullseye** port. It solved the issue for me.

gw6kwm, 2023-10-15:
Same issue coming on the the legacy version
+ pip3 install -U -r /home/birdnet/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.

andrebruton, 2023-10-15:
The Legacy 64 Bit Lite version worked 100%.

gw6kwm, 2023-10-15:
Just did a complete fresh install on a Pi 4
+ pip3 install -U -r /home/birdnet/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.

Tinker4you, 2023-10-16:
Hello, I have the same problem:
++ hash -r
+ pip3 install -U -r /home/bat/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.

Tinker4you, 2023-10-16:
Pi OS (Legacy, 64-bit) Lite version of the Bullseye port. It solved the issue also for me.

joostdjong, 2023-10-17:
> Just did a complete fresh install on a Pi 4
>
> * pip3 install -U -r /home/birdnet/BirdNET-Pi/requirements.txt
>   Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
>   ERROR: [...]

The only thing I can think of is double check you got the right OS-version. There are a couple of Legacy options in the imager. With `cat /etc/os-release` I think you should have a return containing _VERSION="11 (bullseye)"_.

Kaleidox93, 2023-11-09:
This issue has been ongoing, with many prospective solutions proposed, none of which was the panacea. I had been running on Libre 'le potato' AML 950X-cc running their modded Buster, and it had no troubles at all. I opted to upgrade only to see if performance world improve, only to find that installation was not possible given the script bombing. Attempting simple obvious things as we all do netting no joy, various favors of the OS different burning tools, generally all of the things that typically wouldn't fix this sort of problem, but we try them anyway. I considered facing due East while twirling a dead chicken counterclockwise over my head whilst reciting the lords prayer doing the script execution, but opted against it. Occam's Razer became too loud to ignore.

I am moderately Linux savvy, and will take a dedicated look at this over the weekend, in the hopes of seeing something obvious, however - the gurus behind this piece of beloved have exponentially better odds than I do, so please, bringers of BirdNET-Pi, if you have the time, you will make a community of devoted followers have happy.

Best wishes and 73,

Carl

Tamooj, 2023-12-07:
I built with the latest OS version, which includes Py 3.11.  To make it work, I tweaked the BirdNET install to require the latest version of TensorFlow Lite (2.12). I basically replaced reference to version 2.06 with version 2.12, and changed the curl to grab that version locally, from the tensorflow repo (https://github.com/PINTO0309/TensorflowLite-bin/blob/main/2.12.0/download_tflite_runtime-2.12.0-cp311-none-linux_aarch64.whl.sh) and then fed the tweaked install through bash.  The only other issue is the default installation of Apache from the net-tools update, overrides Caddy, so that's the next debug fun.

grleblanc, 2024-04-03:
> I built with the latest OS version, which includes Py 3.11. To make it work, I tweaked the BirdNET install to require the latest version of TensorFlow Lite (2.12). I basically replaced reference to [...]

FYI this worked for me. Thank you!

Willtech, 2024-04-13:
> You could try the Pi OS (**Legacy**, 64-bit) Lite version of the **Bullseye** port. It solved the issue for me.

We don't use Bullseye for anything. We use Buster or Bookworm. At least it works.

HEXIAOMIN888, 2024-12-10:
Are there any new solutions?