mcguirepr89/BirdNET-Pi #1055: Install issue on RPi4

cebelcher, 2023-10-18:
Trying a new install on an 8G Rpi 4 - using  Raspberry Pi OS (Legacy) Lite Bullseye

and the install ends with:

+ pip3 install -U -r /home/cbelcher/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.

Shellfishgene, 2023-10-19:
I just installed successfully, make sure you use the "Raspberry Pi OS (Legacy) Lite Bullseye" image that is "64-bit".

breimann, 2023-10-21:
@Shellfishgene i've done that, but still can't install successfully...

breimann, 2023-10-22:
Any other advice to get this to install correctly?? Im on a 1GB Rpi 4 - using Rpi OS (Legacy, 64bit) Lite bullseye...

mnhn-paul, 2023-11-09:
@breimann did you managed to get it installed? looking at the same problem here...

w2xg, 2023-11-09:
I had the above issue when installing with "Raspberry Pi OS LITE (64-bit)" (as instructed in the Installation Guide) however it worked fine on second attempt when I re-flashed with "Raspberry Pi OS (**Legacy**, 64-bit) Lite".

breimann, 2023-11-09:
No but I haven't tried the April version of bullseye

mnhn-paul, 2023-11-10:
> I had the above issue when installing with "Raspberry Pi OS LITE (64-bit)" (as instructed in the Installation Guide) however it worked fine on second attempt when I re-flashed with "Raspberry Pi OS (**Legacy**, 64-bit) Lite".

did you have to create the "pi" user when flashing the SD card (RaspberryPiImager, need to set a user if I want to enable SSH), if I understand this is no loner create automatically but the BirdNetPi needs it?

breimann, 2023-11-10:
I just created login birder

olemueller, 2023-11-16:
can confirm, the installation on "Raspberry Pi OS LITE (64-bit)" doesn't work atm.

 i used "Raspberry Pi OS (**Legacy,** 64-bit) Lite", worked fine for me.

jeffsikes, 2023-11-30:
> I had the above issue when installing with "Raspberry Pi OS LITE (64-bit)" (as instructed in the Installation Guide) however it worked fine on second attempt when I re-flashed with "Raspberry Pi OS (**Legacy**, 64-bit) Lite".

Thank you! This resolved the issue for me and I was able to install successfully. 🙇🏼

breimann, 2023-12-01:
This didn't help me at all

sarvesh-lad, 2024-02-16:
From what I gather the install_services.sh has php before php-fpm and php requires apache2 which is a new debian version thing

so moving it after php-fpm doesn't install apache:

Are you able to test this:

https://github.com/sarvesh-lad/BirdNET-Pi/

`curl -s https://raw.githubusercontent.com/sarvesh-lad/BirdNET-Pi/main/newinstaller.sh | bash`

Notes on my fork:

1. it assumes 0 as LATTITUDE and LONGITUDE cause that IP-API site is finicky, you can always change that in settings
2. it installs tfruntime directly from pip
3. Has the install services fix I mentioned above.

hugokoopmans, 2025-03-05:
i cannot login to tools with my user, i can ssh in but not in the UI