mcguirepr89/BirdNET-Pi #1056: The installation exited unsuccessfully.... using 64 bit Bullseye, not Bookworm

breimann, 2023-10-22:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
I can't install birdnetpi successfully on my new RPi 4 1GB

**To Reproduce**
Steps to reproduce the behavior:
1. fresh image of "Raspberry Pi OS (Legacy, 64bit) Lite" on 64GB sd card
2. put into Rpi 4 1gb
3. boot up
4. Putty into via ssh and run the install command: curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash
5. I get "The installation excited unsuccessfully" at error:
+ python3 -m venv birdnet
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/home/birder/BirdNET-Pi/birdnet/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']

7. So I manually install python3-venv using "sudo apt-get install python3-venv" 
8. Then use "cd && uninstall.sh && rm -drfv BirdNET* && curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash" to uninstall and reinstall
9. And then i get "The installation exited unsuccessfully" once again, but with:
ERROR: Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Max retries exceeded with url: /packages/47/6a/453160888fab7c6a432a6e25f8afe6256d0d9f2cbd25971021da6491d899/pip-23.3.1-py3-none-any.whl (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1123)')))

The installation exited unsuccessfully.
11. HELP!!!! I JUST WANT TO IDENTIFY OUR BEAUTIFUL BIRDS!!!

**Expected behavior**
A working installation of birdnetpi!

**Your build**
Rpi 4 1GB (brand new), using a 64GB micro sd card

**Code or log snippets**

login as: birder
birder@192.168.1.119's password:
Linux birdnetpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch6                                                                                        4

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

birder@birdnetpi:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
birder@birdnetpi:~ $ uname -a
Linux birdnetpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch6                                                                                        4 GNU/Linux
birder@birdnetpi:~ $ cat /etc/debian_version
12.7
birder@birdnetpi:~ $ python --version
Python 3.9.2
birder@birdnetpi:~ $ curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash
[... 374 lines cut ...]
+ echo 'Adding Caddy sudoers rule'
Adding Caddy sudoers rule
+ cat
+ chmod 0440 /etc/sudoers.d/010_caddy-nopasswd
+ config_icecast
+ '[' -f /etc/icecast2/icecast.xml ']'
+ sed -i 's/>admin</>birdnet</g' /etc/icecast2/icecast.xml
sed: can't read /etc/icecast2/icecast.xml: No such file or directory
+ passwords=("source-" "relay-" "admin-" "master-" "")
+ for i in "${passwords[@]}"
+ sed -i 's/<source-password>.*<\/source-password>/<source-password>birdnetpi<\/source-password>/g' /etc/icecast2/icecast.xml
sed: can't read /etc/icecast2/icecast.xml: No such file or directory
+ for i in "${passwords[@]}"
+ sed -i 's/<relay-password>.*<\/relay-password>/<relay-password>birdnetpi<\/relay-password>/g' /etc/icecast2/icecast.xml
sed: can't read /etc/icecast2/icecast.xml: No such file or directory
+ for i in "${passwords[@]}"
+ sed -i 's/<admin-password>.*<\/admin-password>/<admin-password>birdnetpi<\/admin-password>/g' /etc/icecast2/icecast.xml
sed: can't read /etc/icecast2/icecast.xml: No such file or directory
+ for i in "${passwords[@]}"
+ sed -i 's/<master-password>.*<\/master-password>/<master-password>birdnetpi<\/master-password>/g' /etc/icecast2/icecast.xml
sed: can't read /etc/icecast2/icecast.xml: No such file or directory
+ for i in "${passwords[@]}"
+ sed -i 's/<password>.*<\/password>/<password>birdnetpi<\/password>/g' /etc/icecast2/icecast.xml
sed: can't read /etc/icecast2/icecast.xml: No such file or directory
+ sed -i 's|

ygeshors, 2023-10-22:
Same, impossible to install. It's a shame, because until now it worked very well.

DMontgomery40, 2023-10-25:
I have the same issue, with things taking a turn for the worse at pip3 install, would really appreciate any help here!  Also on Raspberry 4B, 4 GB, brand new, brand new and freshly formatted SD card, flashed with the official Pi imager.  

+ pip3 install -U -r /home/nuthatch/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.

sarvesh-lad, 2023-10-28:
> I have the same issue, with things taking a turn for the worse at pip3 install, would really appreciate any help here! Also on Raspberry 4B, 4 GB, brand new, brand new and freshly formatted SD card, [...]

Check your python version `python -V` on the shell.

it expects python3.9 and you likely have python 3.11.2 installed. 
Bullseye 11.8 comes with 3.11.2 and so does bookworm.

Find an older image of RaspiOS to see if it works.

@breimann Edit this `scripts/install_config.sh`

Comment out the section that find your coordinates and add them in manually upto max 4 decimals:

```
# Retrieve latitude and longitude from web
#json=$(curl -s4 http://ip-api.com/json)
#if [ "$(echo "$json" | jq -r .status)" = "success" ]; then
#  LATITUDE=$(echo "$json" | jq .lat)
#  LONGITUDE=$(echo "$json" | jq .lon)
#else
#  echo -e "\033[33mCouldn't set latitude and longitude automatically, you will need to do this manually from the web #interface by navigating to Tools -> Settings -> Location.\033[0m"
#  LATITUDE=0.0000
#  LONGITUDE=0.0000
#fi

LATITUDE=0.0000
LONGITUDE=0.0000

```

breimann, 2023-10-28:
hi @sarvesh-lad , here's my version of python..

birder@birdnetpi:~ $ python --version
Python 3.9.2

sarvesh-lad, 2023-10-29:
> hi @sarvesh-lad , here's my version of python..
> 
> birder@birdnetpi:~ $ python --version Python 3.9.2

I too am now facing the same -_- gets hung up on installing language model.

```
Requirement already satisfied: smmap<6,>=3.0.1 in ./birdnet/lib/python3.9/site-packages (from gitdb<5,>=4.0.1->gitpython!=3.1.19->streamlit==1.19.0->-r /home/birdnet/BirdNET-Pi/requirements.txt (line 11)) (5.0.1)
Requirement already satisfied: mdurl~=0.1 in ./birdnet/lib/python3.9/site-packages (from markdown-it-py>=2.2.0->rich>=10.11.0->streamlit==1.19.0->-r /home/birdnet/BirdNET-Pi/requirements.txt (line 11)) (0.1.2)
tflite-runtime is already installed with the same version as the provided wheel. Use --force-reinstall to force an installation of the wheel.
+ cd /home/birdnet/BirdNET-Pi/scripts
+ ./install_language_label_nm.sh -l en
Archive:  /home/birdnet/BirdNET-Pi/model/labels_nm.zip
  inflating: /home/birdnet/BirdNET-Pi/model/labels_en.txt
Archive:  /home/birdnet/BirdNET-Pi/model/labels_nm.zip
  inflating: /home/birdnet/BirdNET-Pi/model/labels_en.txt
+ exit 0

```

MickW69, 2023-11-07:
I had this same problem, ended up needing to use [2022-04-04-raspios-bullseye-arm64-lite.img.xz] https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz)

breimann, 2023-11-07:
@MickW69, and did that work for you?

MickW69, 2023-11-07:
Yes, installed successfully, up and running.

breimann, 2023-11-12:
> I had this same problem, ended up needing to use [2022-04-04-raspios-bullseye-arm64-lite.img.xz] https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz)

I just tried doing this using your link to the Rpi os, installed that on my sd, booted up, and ran

curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash

And now i get nothing happening!! At all, like the login prompt just pauses and a new fresh login prompt comes up like it's done nothing with this command

MickW69, 2023-11-16:
About the only other thing I can mention is that each time an OS didn't work I would format and do a fresh write with the Pi imager to try the previous one. So, it took a while to go through the archive of 64bit lite options.

https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit

https://downloads.raspberrypi.com/raspios_lite_arm64/images/?_gl=1*saa9x6*_ga*MTM2MDM0MzAxNC4xNjkzMDI3MTkz*_ga_22FD70LWDS*MTcwMDA5MzMxOC4xLjEuMTcwMDA5MzQxMy4wLjAuMA..

Good luck.

BO-TULIST, 2023-12-15:
Same issue here with current 64 bit lite OS. Birdnet install fails with 

ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.

Installed a fresh copy of Raspberry Pi OS (Legacy, 64 bit) lite, Bullseye released 2023-12-05. 
With that, install worked fine.

einate57, 2024-02-16:
> I had this same problem, ended up needing to use [2022-04-04-raspios-bullseye-arm64-lite.img.xz] https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz)

I had (as many other users) the same issue.
Hardware: brand new PI4
OS: bullseye 12.5 arm lite 64-bit (did the image with PI Imager)
Python version 3.11

The error
`ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.`

appears as TensorFlow needs Python 3.9 to be executed.

Then I use this image [2022-04-04-raspios-bullseye-arm64-lite.img.xz] and follow the instruction again and it worked successfully.

aolsorken, 2024-02-16:
Hi,

When I tried to install BirdNET Pi on a Raspberry Pi model 3, running Raspian OS/Bulls Eye I ran in to the same problem.
Before I started the installation I did check the requirement section in the beginning of installation instruction, but the dependency on Pyhton 3.9 is not listed there.
So my suggestion is that someone involved in the maintenance of BirdNET Pi adds undocumented requirements.