mcguirepr89/BirdNET-Pi #1066: Identifying birds from different regions

dasien, 2023-11-07:
Hello,

After a fresh install of BirdNET-Pi, we are getting detections of birds which are not in our region and none for birds which are.  Prior to this install it was working perfectly.  Now i am getting only birds which are from Africa, Australia, etc.  I have checked my Lat/Lon and they are for our area in NY.  Also checked the Pi to make sure it's region and time zone where set appropriately.  Our egress IP is also correctly showing as our area.

What else, besides the device's location, does the model use to include current location in the rules for identifying birds?  It is 100% incorrect right now (i.e. i see Blue Jays out at our feeder and it has not found one of them.  What it has found, in the attached screenshot, are birds found in Africa and South America).

I have clearly set something up incorrectly, but i can't figure out what it is.  I did have to use a legacy version of the Pi OS as the current one uses a newer version of python than the install script expects, but other than that Birdnet was loaded straight from the installer.

[image]

RoyInAustin, 2023-11-18:
I am having the same problem as dasien with the exception is that I am a new birdnetpi user.  My detections are from S. America region and not Texas.  I do not believe I have had any correct identifications yet.  Plenty of detections and identifications but none for the correct birds I have in my garden or in Texas that I know of. Plenty of audio showing the spectrogram and in the detected bird file base on what I see other systems showing.  All services show running.

Any suggestions appreciated.

RoyInAustin, 2023-11-18:
I am running a Pi 4 B 4GB RAM and 32GB SD.

declanfleming, 2023-11-20:
I haven't seen this, but in Settings, maybe change to the older model and see if there are changes?

RoyInAustin, 2023-11-28:
> I had this problem too.
>
> In the Advanced setting, make sure you haven't altered the privacy threshold. I had, and when I adjusted it back to 1%, I started getting only local region hits.

I got problem fixed.  It was the first options in settings.  I had made no selection initially because I did not think it applied to my setup. All working great now.

Roy Walker

alanhoyle, 2025-02-06:
I was seeing this same issue.  As @RoyInAustin  suggested, the issue with mine was that the default detection model was the 2020 legacy model.  Shifting to the 2023 model immediately improved my results.  

The old model identified a chainsaw as a "Himalayan White-browed Rosefinch" with 95% confidence.  what I believe was a local Wren as a "Bat Hawk" (native to sub-Saharan Africa) with 99% confidence.