mcguirepr89/BirdNET-Pi #1068: Support for BirdNET-Pi on Pi 5 is needed

Inachis, 2023-11-08:
Ref. https://github.com/mcguirepr89/BirdNET-Pi/discussions/1037.

Pi 5 only supports Bookworm, there is currently no possibility to install Bullseye.

BirdNET-Pi installation on Bookworm fails with
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.

Thus Pi 5 is not a viable platform for BirdNET-Pi or BattyBirdNET-Pi.

ostpol, 2023-11-11:
You could try to change the corresponding entry in the requirements.txt file to
`tflite_runtime-2.14.0-cp311-cp311-manylinux_2_34_armv7l.whl` or just `tflite_runtime`

Inachis, 2023-11-17:
An update - changing requirements.txt to tensorflow installs OK. However,  I'm getting **Apache2 Debian Default Page** - somthing's not working in the Caddy install/configuration.

lfpoulain, 2023-11-22:
> An update - changing requirements.txt to tensorflow installs OK. However, I'm getting **Apache2 Debian Default Page** - something's not working in the Caddy install/configuration.
 
It happens to me too !

pddpauw, 2023-12-06:
did anyone manage to resolve this?  I have a few Pi5 which I want to use.  Understand that we could change the txt file, but wondering if Caddy install or config got resolved. thx!

pddpauw, 2023-12-06:
the installation indeed says "successful", but there is an issue with Caddy indeed.  I've compared to a working installation on a PI4 with this PI5 installation:
1) the PI4 does NOT have Apache installed and running, whilst the PI5 does...
During installation, I got this warning:
```
Creating config file /etc/php/8.2/fpm/php.ini with new version
NOTICE: Not enabling PHP 8.2 FPM by default.
NOTICE: To enable PHP 8.2 FPM in Apache2 do:
NOTICE: a2enmod proxy_fcgi setenvif
NOTICE: a2enconf php8.2-fpm
NOTICE: You are seeing this message because you have apache2 package installed.
```
apachectl status give "command not found" on the Pi4, but does give a reply on the Pi5

2) when Caddy is running on the PI5, you will get an error.  This is due to wrong reference in the Caddyfile (/etc/caddy/): it refers to a non existing sock file in the /run/php folder.  Replacing this with php8.2-fpm.sock resolves the error

3) but then, trying to log into the page, I get "file not found", so assuming still somewhere a wrong configuration... (the index.php exists though in the BirdNET-Pi/homepage folder

Any further thoughts on how to resolve this...?

pddpauw, 2023-12-08:
not sure if relevant, but checked the output of getcap:

Pi5 (where I get an error): 
```
getcap -r / 2>/dev/null
/usr/lib/aarch64-linux-gnu/gstreamer1.0/gstreamer-1.0/gst-ptp-helper cap_net_bind_service,cap_net_admin=ep
```
Pi4 (working):
`getcap -r / 2>/dev/null`

rjdavy, 2023-12-13:
In a similar situation with my Pi4 which I wanted to add Birdnet to. I upgraded my OS from buster thinking I'd be fine (Rasp Pi Imager offers bookworm by default).

pddpauw, 2023-12-14:
ok, think I'm getting closer and closer...
1) the permissions are different: there is no write access for the Others group for the files in the BirdNet-Pi folder.  For now, I' manually corrected this.
2) added the handle_logger to the Caddyconfig file, and get below reply:

`403 Forbidden stat /home/pi/BirdSongs/Extracted: permission denied fileserver.(*FileServer).ServeHTTP (staticfiles.go:282) r9uk4n20p`

will look at it again over the weekend most likely

pddpauw, 2023-12-16:
I've resolved the issue.  Will try to post solution and new install script for Bookworm tomorrow

pddpauw, 2023-12-17:
OK, made a new repo for testing purposes on the Pi5, starting from this one.  You can find this here: https://github.com/pddpauw/BirdPi
You can find the changes I had to make in the readme file.  

It would obviously be better if those changes can be included in this repo, so hope someone can own or do this.

Structurally, I believe it would be better and safer if the webpages would be saved in /var/www/ so we could limit access to Caddy user in this case.

Inachis, 2023-12-17:
Just done a clean install on a Pi5 with Bookworm 64bit Lite and the test repo - Installation completed successfully and rebooted but home page (index.php) is not served up, just a empty page, no source, no error. I can get images, e.g. http://192.168.1.139/images/menu.png displays the png.

pddpauw, 2023-12-17:
will do a re-test as we speak, will let you know

Inachis, 2023-12-17:
Just checked in Firefox debugger - getting a 502.

pddpauw, 2023-12-17:
please go to your CaddyFile:
`sudo nano /etc/caddy/CaddyFile`

and replace there:
`php_fastcgi` unix//run/php/php7.4-fpm.sock
by
`  php_fastcgi unix//run/php/php8.2-fpm.sock`

then do:
`sudo systemctl reload caddy`

will update in the installation script.  thanks for testing!!

Inachis, 2023-12-17:
OK , now working but that should have been update **Caddyfile**.

Just been configuring and testing with an old USB PS3 camera/microphone...getting MQTT messages OK.

Thank you!

pddpauw, 2023-12-17:
thanks, and you're right: Caddyfile.

I've further updated the installation script, and should work now.

If this has fully resolved your issue, then please mark as resolved.

Inachis, 2023-12-17:
Resolved!

closed 2023-12-17 by Inachis

natebc, 2024-03-12:
Note: this is only resolved in that the fork works for a Pi 5. The repo where this image is still doesn't work on Pi 5. For anybody coming across this, go see pddpauw's fork: https://github.com/pddpauw/BirdPi