mcguirepr89/BirdNET-Pi #1072: Species Stats page not updating with latest date

pddpauw, 2023-11-13:
the tabsheet "Species Stats" on the PiBirdnet webpage is not updating with the latest available date.  Latest date available, is the day of restart.  In below example; last recordings are till 13-11, but I can only select until 11-11.
Expected behavior would be that the day you can select should equal today.

A restart (in this case full Pi restart) changed the latest available date to the date of the restart, but not resolving the issue.

[image]
[image]

Sysinfo:
[image]

Date of the PI:
[image]

Database has the correct dates:
[image]

Pieter

pddpauw, 2023-11-29:
I've corrected this in the plotly_streamlit.py, but adding st.cache_resource.clear() before connecting to the DB.
So adding this on line 53 (before conn = get_connection(URI_SQLITE_DB)) resolves this issue.
@mcguirepr89 can we add this in the code?

kspearrin, 2024-01-03:
Also seeing this issue. Will try your fix @pddpauw . Thanks!