mcguirepr89/BirdNET-Pi #1080: Password not working everywhere, but does in some places.

MABeatty1978, 2023-12-01:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
I can click on "Tools" and it gives me a prompt to sign in.  I enter my username and password and it takes me into the tools menu.  From the tools menu, I can update settings, system controls, services, custom species list, and excluded lists all without a problem.  However, if I try to select System Info, File Manager, Database Maintenance, or Web Terminal, it prompts me once again for a username and password and the same password I used to get into tools doesn't let me in.  I'm also getting prompted to enter username and password on the spectrogram.

MABeatty1978, 2023-12-01:
I messed around with the for an hour, and for some unknown reason, it started working fine again.  I don't know what I did, if anything that fixed it.  My best guess is that the issue started when I was building a new instance of BirdNetPi on a separate Raspberry Pi and had them both up and running at the same time using the same browser.  I turned the new one off about a half hour ago, so I'm wondering if there was some kind of caching issue?

I'll leave this open if anyone is interested in digging in deeper to figure out what happened, but please feel free to close if not.

Svardsten53, 2024-01-18:
Passwords for different functions are handled in the caddyfile. This makes it possible to set different passwords for different functions.
1. Run the following command and replace $PWD with the password you want users to gain access to, for example, audio listening with: `caddy hash-password --plaintext $PWD`
Copy the result.
 2. Open /etc/caddy/Caddyfile , find the line that reads "basicauth /stream {" and the line below that reads "birdnet _a lot of characters_" Replace the characters with the password you got from step 1 and possibly also change the username.
 3. Run the command: `sudo systemctl reload caddy`