mcguirepr89/BirdNET-Pi #1099: birdnet_analysis.sh stuck with "Maximum number of attempts exceeded. Exiting & restarting service."

zcloud-at, 2024-01-07:
When analyzing many detections, it often happens that the birdnet_analysis.sh script gets stuck in the IF loop on line 107. The Pi must then be restarted.
The restart has to be done manually and unfortunately many analyzes are lost until the error is detected.

Scraggarax, 2024-01-07:
This is something I ran into in error #653. You have to disable the microphone service for the analysis service to catch up and keep refreshing the log.

It seems like getting a larger and/or faster storage might help too.

zcloud-at, 2024-01-07:
@Scraggarax 
Thanks for the tip.
I'm currently using the Pi with two RTSP streams, so there shouldn't be any problems with the microphone input. I'll try with an external USB.3 NVME. Maybe faster memory will help.

Scraggarax, 2024-01-07:
It's less of a microphone issue, and More of the way that the analysis system seems to work. If it records more than it processes, it seems to not be able to keep up, so shutting off the recording service until the analysis process catches up prevents it from getting stuck in the loop. You can kinda see what I mean if you watch the logs when it spits it out.

zcloud-at, 2024-01-09:
Current status:
With the solution of an external NVME storage there are no more failures so far. The files are analyzed much faster and the daily detection rate increases significantly. A clear recommendation not to use an SD card.

zcloud-at, 2024-01-12:
The system got stuck again.

My approach would now be to start the script “BirdNET-Pi/scripts/restart_services.sh” with the listed error message every time in order to get rid of the accumulated incorrect clips and restart the services properly.

closed 2024-01-27 by arxipelagos