mcguirepr89/BirdNET-Pi #1103: BirdNet-Pi Remote View

Phatguppy, 2024-01-12:
Is there any good documentation on getting BirdNet-Pi up and running? I have the correct Raspberry Pi OS installed, BirdNet-Pi installed on a model 4b is running, I am logged in on the Pi. But I cant access the web page from a remote computer, and I cant assess if its my router blocking .local or if its an incorrect setting on BirdNet-Pi to serve the web page. I am stuck.

pddpauw, 2024-01-12:
this is a too brief description of the issue to help to resolve it...
The main things which need to happen, is 1) ensure you have the correct re-direct in your router to your Pi, 2) have a static public IP or DNS.
I really can't tell you much more, but if you want to access remotely, you will need to the public IP assigned to your router...

Alternatively, you might want to work with a VPN connection, so both your client device (you want to use to access this Pi) and your Pi are on the same VPN.  You can install a VPN server (e.g. OpenVPN) behind your router.

Please give us more info on i) what you want to achieve, ii) what your network configuration is and iii) what you've tried already.

Phatguppy, 2024-01-13:
Sorry for the lack of context, my computer and Pi are all on my local network. I fixed an error on my mac preventing me to ssh to the pi, I can now do that. Questions:

Can I determine that my Pi is running BirdNet-pi correctly from the remote client? Any command to test the software? 
In the latest install of BirdNet-Pi, are there any other fixes that I need to do that might not be in the instructions?

pddpauw, 2024-01-13:
hi, I'm not sure I fully understand.  After your initial post, I thought you meant with "remote" from outside your local network, but reading above, I think you mean "another computer in the home network".

You have a couple of possibilities:
1) try (e.g. on your laptop) to go to the site via the ip nr. For this, you would need to know the IP of your Pi.  After logging in via SSH, you can type ifconfig as command in the terminal, and this will give you the IP.  Then go to http://theIPofyourPi.
I would assume you can ping from your laptop to the IP of your Pi.

2) alternatively, you can check if everything is up and running on your pi: once logged in via SSH, you can do:
`curl 127.0.0.1`
as output, you should get the ini file in your terminal window

if you can't access, check if Caddy (=webserver) is running:
`sudo systemctl status caddy`

3) to check if BirdAnalyzer is running (which is different from your webserver running (!)), you can do 
` sudo systemctl status birdnet_server.service`
if needed, you can restart with
` sudo systemctl restart birdnet_server.service`

hope this gives a bit of additional help.  In case you're still stuck, please give the output of:
`sudo systemctl status caddy`
`curl 127.0.0.1`

Phatguppy, 2024-01-21:
Hi all, I wanted to follow-up on my posts above. I had some local network issues that are still unresolved so what I did was a fresh install of Debian Bullseye 64bit with the desktop OS and apps. Now, my BirdNet-Pi is running just fine, and I can control everything from my Pi, which already had a monitor and keyboard. Thanks for the above post for debugging, that helped!