mcguirepr89/BirdNET-Pi #1111: Further Installation Issues

alecjcook, 2024-01-22:
The following steps were taken to install birdnet-pi

- Installer the latest Raspbien Lite OS-64 on USB using the Raspberry Pi Imager v1.8.4
- Used the raspi-config to expand the file system and rebooted
- `sudo apt-get update`
- `sudo apt-get upgrade`
- `sudo apt-get install git jq`
- run `curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash`

The installation process ends thus:
```
+ pip3 install -U -r /home/birder/BirdNET-Pi/requirements.txt
ERROR: Invalid requirement: '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' (from line 1 of /home/birder/BirdNET-Pi/requirements.txt)
The installation exited unsuccessfully.
```

I Have attaced a full terminal printout of the installation process

[MobaXterm_birdnetpi1birder_20240122_141624.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/14011240/MobaXterm_birdnetpi1birder_20240122_141624.txt)

stevedee, 2024-01-22:
I'm pretty sure you still need Debian "Bullseye" (Legacy) not the latest OS version.

alecjcook, 2024-01-22:
https://github.com/mcguirepr89/BirdNET-Pi/issues/1049
Install OS "PI OS Legacy 64-Bit"
This installation guide is out of date...

Svardsten53, 2024-01-28:
I just tried an installation on a Raspberry Pi 4B 8GB without any problems. I guess it's important what version of OS you are using. This file works: **2023-05-03-raspios-bullseye-arm64-lite.img.xz**
You must specifically select the file in the Raspberry Pi Imager.

Svardsten53, 2024-01-28:
It seems that using Bullseye OS is not enough, it has to be 64-bit light and probably this has changed a few times so this is causing your error. Please see what I wrote in issue #1111

If you just want to have a new working installation you can take an image copy of the working one and restore the image to a new card and change data as latitude, longitude, password etc. Don't forget to clear all data to have an empty database.

sarvesh-lad, 2024-02-16:
From what I gather the install_services.sh has php before php-fpm and php requires apache2 which is a new debian version thing

so moving it after php-fpm doesn't install apache:

Are you able to test this:

https://github.com/sarvesh-lad/BirdNET-Pi/

`curl -s https://raw.githubusercontent.com/sarvesh-lad/BirdNET-Pi/main/newinstaller.sh | bash`

Notes on my fork:

1. it assumes 0 as LATTITUDE and LONGITUDE cause that IP-API site is finicky, you can always change that in settings
2. it installs tfruntime directly from pip
3. Has the install services fix I mentioned above.

alexbelgium, 2024-05-16:
There's is a new fork solving installation issues here : https://github.com/Nachtzuster/BirdNET-Pi