mcguirepr89/BirdNET-Pi #1118: Install issue (yes, another one of these)

YetiOfTheSky, 2024-01-28:
I'm having the similar issues as a few others.

I've already got one of my Pi's running BridNet-Pi - it has been running perfectly for about a year.
Setting up another for a friend but it keeps failing on install.
I had seen a few posts in here about making sure you went with the older version of OS and check the python version.

So I matched the OS and python ver to my current successful running one but I'm still getting install fails :(
This is the error I get..
[image]

I have double-checked against my working Pi install and the Reference, lsb_release, and python version all match up.
I don't really have enough Linux skill to work out where it's falling over.
Hoping someone can look at the above screenshot and give me a clue?

Svardsten53, 2024-01-28:
It seems that using Bullseye OS is not enough, it has to be 64-bit light and probably this has changed a few times so this is causing your error. Please see what I wrote in issue #1111

If you just want to have a new working installation you can take an image copy of the working one and restore the image to a new card and change data as latitude, longitude, password etc. Don't forget to clear all data to have an empty database.

YetiOfTheSky, 2024-01-29:
Hi, and thanks for the reply.

Both the Pi's I have are running 2023-05-03-raspios-bullseye-arm64-lite.img.xz
This is what is confusing me.
I'm not sure how the one I set up initially works fine, but the one I'm setting up now, using the same image and the same install process, keeps failing.

But thanks for the suggestion about cloning my working one, I did not consider that, and it's probably worth a try as I'm not getting anywhere with the normal install approach.

spaceof7, 2024-02-02:
I got the same "index-pack failed" error while trying to install as well. The error looked like it was caused by a network problem so I ran the last 2 commands manually, which completed the install for me.

```
git clone -b main --depth=1 https://github.com/mcguirepr89/BirdNET-Pi.git ~/BirdNET-Pi
~/BirdNET-Pi/scripts/install_birdnet.sh
```

YetiOfTheSky, 2024-02-02:
Nice one, thanks very much spaveof7, I'll give that a try later today.

YetiOfTheSky, 2024-02-04:
Hi spaceof7 - thanks for the advice.
Your extra commands got my second Pi up and running!
Thanks very much for your help, really appreciated :)

closed 2024-02-04 by YetiOfTheSky