mcguirepr89/BirdNET-Pi #1122: Caddy Setup

rsympson, 2024-01-30:
I've been running 2 BirdNET stations for a year now.  One of them got some corrupted files so I've rebuilt it.  The latest build results in a Caddy screen asking for for some setup steps that I don't understand.  Are there some clear step by step instructions that will get me past this Caddy interuption and back to listening?

rsympson, 2024-01-31:
I've done some additional rebuilds, making sure to use the RPi OS Lite 64 bit Bullseye version and am getting the same results.  On my desktop computer to web page that renders is an "Apache2 DebianDefault Page".  What has changes wiht the build process such that the green BirdnetPi Green overview page no longer renders after a successful build?

[image]

rsympson, 2024-01-31:
This is the Caddy page I was presented with as referred to in my first post.

[image]

pddpauw, 2024-02-09:
I had similar issues when trying to make this work on Bookworm (Pi5).
I closed apache:

`sudo systemctl disable apache2 && sudo systemctl stop apache2`

and enabled caddy (incl starting at boot): 
`sudo systemctl enable caddy && sudo systemctl start caddy`

I also hade some issue with user rights as Caddy couldn't write to all of the files, so made some chmod changes in the /home/pi folder

rsympson, 2024-02-09:
Thanks.  I'll try a new build and see if I can get it to work.  My current workaround was to take an image from my second BirdNET setup, restore it to my failed one, and make the necessary name and network changes.  This is working fine, but only because I had a functioning second system.

BS

sarvesh-lad, 2024-02-16:
From what I gather the install_services.sh has php before php-fpm and php requires apache2

so moving it after php-fpm doesnt install apache:

Are you able to test this:

https://github.com/sarvesh-lad/BirdNET-Pi/

`curl -s https://raw.githubusercontent.com/sarvesh-lad/BirdNET-Pi/main/newinstaller.sh | bash`

Notes on my fork:

1. it assumes 0 as LATTITUDE and LONGITUDE
2. it installs tfruntime directly from raspberypi
3. Has the install services fix I mentioned above.