mcguirepr89/BirdNET-Pi #1131: Mounted Disk Space Full - BirdNET Pi System Failure on Raspberry Pi 4b

Rebero, 2024-02-15:
## Environment
- **Device**: Raspberry Pi 4 B (PCB 1.5 Sony UK)
- **OS**: Debian GNU/Linux 11 (bullseye)
- **BirdNET Pi Version**: [Please insert the version if known]

## Description
After running smoothly for a couple of days, my BirdNET Pi installation on Raspberry Pi 4b encounters an error where it fails due to the mounted disk space running full.

## Expected Behavior
BirdNET Pi should run continuously without encountering disk space issues, or it should manage disk space appropriately to prevent system failure.

## Actual Behavior
The system logs indicate repeated failures with the message:

/etc/birdnet/birdnet.conf: line 31: Y95SwLUM3tqmnhZ9wuGqXzTE: command not found

This suggests that there might be an issue with the configuration or a script that's causing the system to attempt executing a non-existent command, which could be related to the disk space issue.

## Steps to Reproduce
1. Install BirdNET Pi on Raspberry Pi 4b with Ubuntu Bullseye 64-bit lite.
2. Let it run for a couple of days.

## Log Files and Screenshots
- The log activity shows constant failures related to the `birdnet.conf` file.
- The system information screenshot indicates that the disk space is completely
[image]
[image]
 full.

## Additional Context
- The system uptime was 1 day, 22 hours, and 20 minutes before the last boot.
- Physical memory usage is at 19% with a total of 1.48 GiB used out of 7.63 GiB.
- The `/dev/root` partition is at 100% usage with 0 B free out of 112.94 GiB.

I would appreciate any guidance on how to resolve this issue or any potential workarounds to prevent this from happening in the future.

Thank you for your assistance.

Nachtzuster, 2024-02-19:
@Rebero the 'command' that is not found here is a string you do not want to leak I think. Better remove it from your text and blur out the string in your screenshot.

Rebero, 2024-02-20:
@Nachtzuster Thank you for the hint. I have updated the text and screenshot.

PeterMescher, 2024-02-22:
I'm assuming that under "Advanced Settings" you've set the "Full Disk Behavior" to "purge"?

Nachtzuster, 2024-02-23:
Most likely the disk ran full because analysis is not happening.
The log shows two different errors that would indicate that the analysis pipeline is crashing.

I wonder if the config file is corrupted, the errors hint of that. 
Could you post `/etc/birdnet/birdnet.conf`

Rebero, 2024-02-23:
@PeterMescher  Thx for the follow up. Yes, I did.