mcguirepr89/BirdNET-Pi #1140: Apprise $flickrimage field is not being passed appropriately

hestonhamilton, 2024-02-27:
**Describe the bug**
Use of the "$flickrimage" variable in the Apprise notification field results in an image of a penguin being returned for all detected species.

I have personally experienced this bug while using Discord webhook integrations but there are recent reports of this issue occurring in MQTT implementations as well:

https://github.com/mcguirepr89/BirdNET-Pi/discussions/342
https://tatageek.blog/2024/02/07/rozpoznani-hlasu-ptaku-pomoci-umele-inteligence/
 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to Tools > Settings > Notifications in the BirdNET Pi web interface
2. Set Flickr API Key in "Flickr API Key" field under "Bird Photos from Flickr" section.
3. Paste Discord webhook URL in "Apprise Notifications Configuration" field.
4. Use "$flickrimage" anywhere in the "Notification Body" field.
5. Penguin photo below is returned in the Apprise notification for all detected species.

[image]

**Expected behavior**
Use of "$flickrimage" in the Apprise "Notification Body" field should return a photo that matches the detected species.

**Screenshots**

BirdNET Pi settings -

[image]

Discord webhook post -

[image]

**Additional context**
Flickr image retrieval works fine everywhere on the web interface, but Apprise Flickr image retrieval seems to be broken.

**Your build**
Raspberry Pi 4

TheGingeraffe, 2024-03-03:
I was wrong. While this behavior may have been the result of the species not being passed appropriately on test notifications, it seems that everything on the Flickr side works just fine for actual detections.

My only suggestion is to perhaps change the "send test notification" box's command to include the correct species rather than the penguin image for sanity's sake.

[image]

hestonhamilton, 2024-03-03:
I was wrong. While this behavior may have been the result of the species not being passed appropriately on test notifications, it seems that everything on the Flickr side works just fine for actual detections.

My only suggestion is to perhaps change the "send test notification" box's command to include the correct species rather than the penguin image for sanity's sake.

[image]

Thank you for the wonderful project!

closed 2024-03-03 by hestonhamilton