mcguirepr89/BirdNET-Pi #1142: Installation failed (Caddy)

PracaRaphael, 2024-03-01:
Bonjour les amis,
j'ai un souci, l'installation se passe bien sur mon raspberry, (PI4 debian 64), mais l'url ne delivre que une page "CADDY congratulation etc..."

je ne vois riendans la log et n'ai plus d'idée? 

pouvez vous m'aider?

Merci

caschulz88, 2024-04-17:
What my friend mean is that the installation went through perfectly fine but instead of bird net-pi software just some Caddy server congratulations page is being shown to the user.

I have exactly the same issue on my installation. Following the install guide with no errors popping up.

Can anybody help or give a hint where to check?! Thnx!

caschulz88, 2024-04-18:
I tried it again with bullseye instead of bookworm raspios as base system but still same error. It ended up with a message saying that the installation was completed successfully. But on http server there is only this Caddy message 😞

caschulz88, 2024-04-20:
Seems like the error came from a VERY long loading of the birdnet pi app on the web server. I used the Raspi zero 2 w and will now switch to something which has more power as the zero is absolutely not working! You should remove it from the possible architectures as with 512MB RAM it takes like 4hours to boot up BirdNet PI and everything is so unusable slow.