mcguirepr89/BirdNET-Pi #1146: Installation fails due to inability to create TLS connections to www.piwheels.org

gkenaley, 2024-03-04:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
Installation fails due to inability to create TLS connections to www.piwheels.org. Site seems to be down/flaky, others have noted.
Installation log (attached) unable to get several WHL files from piwheels.org and finally fails with a related python stack dump:

File "/usr/share/python-wheels/urllib3-1.26.5-py2.py3-none-any.whl/urllib3/response.py", line 443, in _error_catcher
    raise ReadTimeoutError(self._pool, None, "Read timed out.")
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Bullseye 64 bit Raspbian Lite on 32 GB SD card on 8GB RPI4 model B.
2. Install as per https://github.com/mcguirepr89/BirdNET-Pi/wiki/Installation-Guide
3. Note I am using an HDMI HD monitor connected directly to the RPI4, not SSH.
4. curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash

**Expected behavior**
Successful Install

**Additional context**
I am a software engineer/developer.

**Your build**
 Bullseye 64 bit Raspbian Lite on 32 GB SD card on 8GB RPI4 model B.

**Code or log snippets**

Setting up php-sqlite3 (2:7.4+76) ...
Setting up icecast2 (2.4.4-4) ...
update-rc.d: error: no runlevel symlinks to modify, aborting!
Setting up libavutil56:arm64 (8:4.3.6-0+deb11u1+rpt5) ...

Collecting pytest-mock==3.7.0
  Downloading https://www.piwheels.org/simple/pytest-mock/pytest_mock-3.7.0-py3-none-any.whl (12 kB)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/suntime/
Collecting suntime
  Downloading https://www.piwheels.org/simple/suntime/suntime-1.2.5-py3-none-any.whl (7.2 kB)

 Downloading https://www.piwheels.org/simple/markdown/Markdown-3.5.2-py3-none-any.whl (103 kB)
Collecting click>=5.0
  Downloading https://www.piwheels.org/simple/click/click-8.1.7-py3-none-any.whl (97 kB)
Collecting packaging
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/packaging/packaging-23.2-py3-none-any.whl
  Downloading https://www.piwheels.org/simple/packaging/packaging-23.2-py3-none-any.whl (53 kB)
Collecting attrs>=19.2.0

Collecting typing-extensions>=3.10.0.0
  Downloading https://www.piwheels.org/simple/typing-extensions/typing_extensions-4.10.0-py3-none-any.whl (33 kB)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/watchdog/
Collecting watchdog
  Downloading watchdog-4.0.0-py3-none-manylinux2014_aarch64.whl (82 kB)

Collecting seaborn
  Downloading https://www.piwheels.org/simple/seaborn/seaborn-0.13.1-py3-none-any.whl (294 kB)
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/share/python-wheels/urllib3-1.26.5-py2.py3-none-any.whl/urllib3/response.py", line 438, in _error_catcher
    yield
  File "/usr/share/python-wheels/urllib3-1.26.5-py2.py3-none-any.whl/urllib3/response.py", line 519, in read
    data = self._fp.read(amt) if not fp_closed else b""
  File "/usr/share/python-wheels/CacheControl-0.12.6-py2.py3-none-any.whl/cachecontrol/filewrapper.py", line 62, in read
    data = self.__fp.read(amt)
  File "/usr/lib/python3.9/http/client.py", line 458, in read
    n = self.readinto(b)
  File "/usr/lib/python3.9/http/client.py", line 502, in readinto
    n = self.fp.readinto(b)
  File "/usr/lib/python3.9/socket.py", line 704, in readinto
    return self._sock.recv_into(b)
  File "/usr/lib/python3.9/ssl.py", line 1241, in recv_into
    return self.read(nbytes, buffer)
  File "/usr/lib/python3.9/ssl.py", line 1099, in read
    return self._sslobj.read(len, buffer)
socket.timeout: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/cli/base_command.py", line 223, in _main
    status = self.run(options, args)
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/cli/req_command.py", line 180, in wrapper
    return func(self, options, args)
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/commands/install.py", line 346, in run
    requirement_set = resolver.resolve(
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 122, in resolve
    self._result = resolver.resolve(
  File "/usr/share/python-wheels/resolvelib-0.5.4-py2.py3-none-any.whl/resolvelib/resolvers.py", line 453, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
  File "/usr/share/python-wheels/resolvelib-0.5.4-py2.py3-none-any.whl/resolvelib/resolvers.py", line 347, in resolve
    failure_causes = self._attempt_to_pin_criterion(name, criterion)
  File "/usr/share/python-wheels/resolvelib-0.5.4-py2.py3-none-any.whl/resolvelib/resolvers.py", line 205, in _attempt_to_pin_criterion
    for candidate in criterion.candidates:
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/factory.py", line 239, in iter_index_candidates
    candidate = self._make_candidate_from_link(
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/factory.py", line 167, in _make_candidate_from_link
    self._link_candidate_cache[link] = LinkCandidate(
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 296, in __init__
    super(LinkCandidate, self).__init__(
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 144, in __init__
    self.dist = self._prepare()
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 222, in _prepare
    dist = self._prepare_distribution()
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 307, in _prepare_distribution
    return self._factory.preparer.prepare_linked_requirement(
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/operations/prepare.py", line 480, in prepare_linked_requirement
    return self._prepare_linked_requirement(req, parallel_builds)
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/operations/prepare.py", line 503, in _prepare_linked_requirement
    local_file = unpack_url(
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/operations/prepare.py", line 253, in unpack_url
    file = get_http_url(
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/operations/prepare.py", line 130, in get_http_url
    from_path, content_type = download(link, temp_dir.path)
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/network/download.py", line 163, in __call__
    for chunk in chunks:
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/cli/progress_bars.py", line 168, in iter
    for x in it:
  File "/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/pip/_internal/network/utils.py", line 64, in response_chunks
    for chunk in response.raw.stream(
  File "/usr/share/python-wheels/urllib3-1.26.5-py2.py3-none-any.whl/urllib3/response.py", line 576, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/share/python-wheels/urllib3-1.26.5-py2.py3-none-any.whl/urllib3/response.py", line 541, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "/usr/lib/python3.9/contextlib.py", line 135, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/share/python-wheels/urllib3-1.26.5-py2.py3-none-any.whl/urllib3/response.py", line 443, in _error_catcher
    raise ReadTimeoutError(self._pool, None, "Read timed out.")
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out.

[installation-2024-03-04.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/14486823/installation-2024-03-04.txt)

gkenaley, 2024-03-04:
I deleted the aborted birdnet-pi installation, rebooted the RPI4 and installed again.
This time it worked fine and I am up and running.
Note that I did not change ANYTHING else.
So it seems to me the problem is or was:
1. Poor Wi-Fi 2.4 GHz connection (unlikely; I am 20 feet from the router through one interior wall)? Perhaps the installer should have some sort of network confidence test added?
2. Intermittent/poor connection to piwheel.org? Originally I thought this was the case because when I googled for piwheel.org I found such complaints.
3. Rebooting the RPI4 made some upgrades installed by the previous failed installations active and thus made the second installation succeed.
So I would not be opposed to closing this defect, although someone else may very well hit it.
Now if I just knew why birdnet-pi is detecting Cuban and Columbian birds in San Diego...

xreec, 2024-04-07:
You can change the Species Occurence Frequency Threshold under settings (use lon/lat) and/or try the other model.