mcguirepr89/BirdNET-Pi #1152: Species stats not working rpi4

yarrow09, 2024-03-08:
Hello,
I discovered the fantastic birdnetpi project on my raspberry pi 4.
I encounter a problem on the "species stats" page which remains completely empty
yet streamlit seems to work:
Top command:
`91304 arnaud    20   0  640584 159960  66428 R 105.6   4.1   0:06.60 python3
 691322 arnaud    20   0  423568 139332  49044 R  99.7   3.6   0:05.45 streamlit
 687246 arnaud    20   0  965652 404472 156308 S  66.6  10.4   1:06.45 python3
 687269 arnaud    20   0  567356  49448  39164 R  17.5   1.3   1:00.23 ffmpeg
   1151 arnaud     9 -11  946840  24792  18256 S   7.3   0.6  54:49.43 `

Status birdnet stats :
`BirdNET-Pi$ sudo service birdnet_stats status
● birdnet_stats.service - BirdNET Stats
     Loaded: loaded (/home/arnaud/BirdNET-Pi/templates/birdnet_stats.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2024-03-08 15:45:33 CET; 972ms ago
   Main PID: 692537 (streamlit)
      Tasks: 4 (limit: 3933)
        CPU: 1.176s
     CGroup: /system.slice/birdnet_stats.service
             └─692537 /home/arnaud/BirdNET-Pi/birdnet/bin/python3 /home/arnaud/BirdNET-Pi/birdnet/bin/streamlit run /home/arnaud/BirdNET-Pi/scripts/plotly_streamlit.py --browser.gatherUsageStats false --serve`

`birdnet:~/BirdNET-Pi$ sudo service birdnet_stats status
● birdnet_stats.service - BirdNET Stats
     Loaded: loaded (/home/arnaud/BirdNET-Pi/templates/birdnet_stats.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Fri 2024-03-08 15:48:49 CET; 3s ago
    Process: 694520 ExecStart=/home/arnaud/BirdNET-Pi/birdnet/bin/streamlit run /home/arnaud/BirdNET-Pi/scripts/plotly_streamlit.py --browser.gatherUsageStats false --server.address localhost --server.baseUrlPath /stats (code=exited, status=1/FAILURE)
   Main PID: 694520 (code=exited, status=1/FAILURE)
        CPU: 4.272s`

thank you in advance for your help, I hope my description of the problem is clear :)

davidalisterbell, 2024-03-25:
I'm having this same issue also on RP4. Any luck on figuring it out?

davidalisterbell, 2024-03-25:
I fixed it on my end, just had to go to tools->services->Streamlit Statistics->restart