mcguirepr89/BirdNET-Pi #1158: Question: What happens during Purge Operation?

Chewie9999, 2024-03-11:
Is it just the mp3 files that are deleted?
If I copied off the mp3s files before the purge happened (I have limited space on my Pi), can I somehow link them up with the detections in the database some time in the future?

wokkeltje13, 2024-03-18:
for me "Purge" is doing nothing, the disk is just running full if I don't remove all data once in a while. Could this be an issue?

MueJosh, 2024-04-04:
This topic fits better to "discussion", it's not an issue

> Is it just the mp3 files that are deleted?

#### Yes

> If I copied off the mp3s files before the purge happened (I have limited space on my Pi), can I somehow link them up with the detections in the database some time in the future?

The birds.db doesn't get purged, you can still see the detections (and other information)- but not the audio file. 
If you copy the audio files/folder to the intended directory (on your pi) you can see and listen to them again. 

It's also possible on your pc with SQLite, but I don't know how

Chewie9999, 2024-04-04:
> This topic fits better to "discussion", it's not an issue

True, although it would be nice if the documentation is updated to explain this.
Thanks for the reply!

MueJosh, 2024-04-04:
This repo doesn't get any updates for now... 

If you want a fork that's still maintained I can recommend the birdnetpi fork from @Nachtzuster

Chewie9999, 2024-04-04:
> This repo doesn't get any updates for now...

Oh :(

> If you want a fork that's still maintained I can recommend the birdnetpi fork from @Nachtzuster

Thank you, I will check it out!

wokkeltje13, 2024-04-05:
can you help me how to modify to a new fork? or do I need to do a fresh install?

MueJosh, 2024-04-05:
> can you help me how to modify to a new fork? or do I need to do a fresh install?

There is no need for a fresh install 
Check out [migrating to nachtzuster fork](https://github.com/Nachtzuster/BirdNET-Pi?tab=readme-ov-file#migrating)