mcguirepr89/BirdNET-Pi #1159: Wiki Addition - Post your daily charts to Mastodon

Chewie9999, 2024-03-11:
I have no idea if this is any use to anyone, but I made a quick script to upload the daily Top 10 chart to a bot on my Mastodon server:

1. It's probably best to create a specific account and flag it as a bot on your mastodon server, so people can mute it specifically, rather than muting your normal account.
2. Create an API key by going to Development -> New Application, and give it "write:media" and  "write:statuses" permissions
3. Save the "Your access token" string into a file on your Birdnet PI server (under `'/home/birdnet/BirdNET-Pi/token.secret`)
4. Save the below code into `/home/birdnet/BirdNET-Pi/mastodonpost.py` or similar
5. Make the file executable using `chmod +x /home/birdnet/BirdNET-Pi/mastodonpost.py`
6. Use `crontab -e` to edit the scheduler and run the command every day at 1minute past midnight:
```
1 0 * * * /home/birdnet/BirdNET-Pi/mastodonpost.py
```

Python Script `mastodonpost.py`
```
#!/bin/python
from mastodon import Mastodon
from datetime import date
from datetime import timedelta

#   Set up Mastodon
mastodon = Mastodon(
    access_token = '/home/birdnet/BirdNET-Pi/token.secret',
    api_base_url = 'https://MASTODONSERVER'
)

# Get current date
today=date.today()
yesterday = today - timedelta(days = 1)
yesterdaystr= yesterday.strftime('%Y-%m-%d')
chartfilename= "/home/birdnet/BirdSongs/Extracted/Charts/Combo-" + yesterdaystr + ".png"
print (yesterdaystr)
print (chartfilename)
# Post message
media = mastodon.media_post(chartfilename, description="Today's top 10 bird chart")
mastodon.status_post("#Birdnet #OtherHashTags",media_ids=media)
```

rabede, 2024-03-20:
Thank you! @BirdnetPi_Leverkusen@mastodon.social is using it now.

magnetron01123, 2024-04-30:
Cool idea. But unfortunately it doesn't work for me. I have followed the instructions and adapted “api_base_url”. 
I also had to adjust the paths slightly.
The error message is: 

```
Traceback (most recent call last):
  File “/home/pi/BirdNET-Pi/mastodonpost.py”, line 2, in <module>
    from mastodon import Mastodon
ImportError: No module named mastodon
```

Does anyone know what I am doing wrong? Do I need to install another module?
Thank you!

Chewie9999, 2024-05-03:
> Traceback (most recent call last):
>   File “/home/pi/BirdNET-Pi/mastodonpost.py”, line 2, in <module>
>     from mastodon import Mastodon
> ImportError: No module named mastodon
> ```
> 
> Does anyone know what I am doing wrong? Do I need to install another module? Thank you!

Oops, Sorry, i missed out an important first step:
```
pip3 install Mastodon.py
```
:(

magnetron01123, 2024-05-04:
> > Traceback (most recent call last):
> > File “/home/pi/BirdNET-Pi/mastodonpost.py”, line 2, in 
> > from mastodon import Mastodon
> > ImportError: No module named mastodon
> > ```
> > 
> > Does anyone know what I am doing wrong? Do I need to install another module? Thank you!
> > ```
> 
> Oops, Sorry, i missed out an important first step:
> 
> ```
> pip3 install Mastodon.py
> ```
> 
> :(

Okay, thanks. That's what ChatGPT told me too. :-) Unfortunately I still have the same error message. Do I have to configure smothering there as well? Thank you for your answer!