mcguirepr89/BirdNET-Pi #1161: Installation exited unsuccessfully (RPC failed - download issue?)

AlaskanAstro, 2024-03-13:
**Describe the bug**
New installation has failed several times at the same place each time. Following the 

**To Reproduce**
Steps to reproduce the behavior:
1. Install fresh image of Raspi Lite 64bit (I tried both Bookworm and Bullseye and failed at same point)
2. Follow guide to access via Putty SSH
3. copy install script and hit enter
4. install proceeds for several minutes
5. install stops with error message

**Expected behavior**
listen to birds

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Tried over wifi and hardline ethernet. No noticeable internet outages. Always fails at this stage, usually past 80MiB: 
```
Receiving objects:  31% (42/132), 18.04 MiB | 1.78 MiB/s
```
**Your build**
Raspi 4B 4GB

**Code or log snippets**
```
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u8) ...
Cloning into '/home/nina/BirdNET-Pi'...
remote: Enumerating objects: 132, done.
remote: Counting objects: 100% (132/132), done.
remote: Compressing objects: 100% (123/123), done.
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8                    )
error: 2602 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: index-pack failed
The installation exited unsuccessfully.
```

AlaskanAstro, 2024-03-13:
Disregard? I kept trying over and over and on the 20th attempt it got through. Might be my own internet?

svetlemodry, 2024-04-03:
I've seen the same error message on my install attempt today shortly ago. Took me just 3 retries to get over this one.

alexbelgium, 2024-05-16:
Hi, use this fork which is the latest updated one : https://github.com/Nachtzuster/BirdNET-Pi