mcguirepr89/BirdNET-Pi #1167: Questions about restarting core services

sdaltons1, 2024-03-18:
I have birdnet up and running on a Pi 400 listening to the audio from two rtsp cameras. It's mostly working great but I notice every couple hours the bird detections stop and a quick restart of core services gets me going again. 

I'm going out of town tomorrow so I can't investigate other solutions at the moment. So 2 questions:

1. Is it possible to restart core services over ssh?
2. Is it possible to write a cron job that would do this automatically every 3-4 hours?

If someone could just tell me what each of those things would look like I would appreciate it!!

Svardsten53, 2024-03-18:
1. You can use command `restart_services.sh`

2. Edit crontab by command `crontab -e`
    Example of command to add in crontab to restart every 4th hour: `0 */4 * * * sudo restart_services.sh`

sdaltons1, 2024-03-18:
> 1. You can use command `restart_services.sh`
> 2. Edit crontab by command `crontab -e`
>    Example of command to add in crontab to restart every 4th hour: `0 */4 * * * sudo restart_services.sh`

Perfect, thank you!

closed 2024-03-18 by sdaltons1