mcguirepr89/BirdNET-Pi #1180: Expected sha256 6646d3f3a256d79c69d2264fa71100dad04f3b42cfb0b203f45d1948bad91f3b

Willtech, 2024-04-13:
Using Raspberry Pi Zero 2W 

Installer says,

```log BirdNet Installer
 Downloading https://www.piwheels.org/simple/streamlit/streamlit-1.19.0-py3-none-any.whl (9.6 MB)
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    streamlit==1.19.0 from https://www.piwheels.org/simple/streamlit/streamlit-1.19.0-py3-none-any.whl#sha256=6646d3f3a256d79c69d2264fa71100dad04f3b42cfb0b203f45d1948bad91f3b (from -r /home/admin/BirdNET-Pi/requirements.txt (line 11)):
        Expected sha256 6646d3f3a256d79c69d2264fa71100dad04f3b42cfb0b203f45d1948bad91f3b
             Got        5cbbc44328a9a344af14840c25d4cd2bb30ba5530b1e0d0aad06a73bf9523a17

The installation exited unsuccessfully.
```

Where can I download a working .img of BirdNET for Raspberry Pi Zero 2W I don't care about stuffy installers I can expand the filesystem myself.

MueJosh, 2024-04-14:
I've never had issues installing and running BirdNET-Pi on any 02W 

If you'd like a faster version of birdnet-pi you should check out the fork from nachtzuster. 
It works much better than this repo (on 02w)

Lotwook, 2024-04-16:
I get this error as well. Installing on a P4 4GB using Legacy Bullseye Lite OS image as base. My Got hash is different though..