mcguirepr89/BirdNET-Pi #1186: Web interface is not reachable after Linux Apt Updates (Bullseye) for PHP

tomtom215, 2024-04-21:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**

After running the usually `sudo apt update && sudo apt upgrade` (which only showed mainly PHP updates) my birdNET-Pi web interface is not reachable. This install has been running rock solid for over a year and I always applied rolling patches when needed. 

**Expected behavior**
BirdNetPi behaves normally after normal linux apt package updates

**Additional context**
Currently installed PHP apt packages:

php-xml/oldstable,oldstable,now 2:7.4+76 all [installed]
php-zip/oldstable,oldstable,now 2:7.4+76 all [installed]
php7.4-cli/oldstable-security,now 7.4.33-1+deb11u5 arm64 [installed,automatic]
php7.4-common/oldstable-security,now 7.4.33-1+deb11u5 arm64 [installed,automatic]
php7.4-curl/oldstable-security,now 7.4.33-1+deb11u5 arm64 [installed,automatic]
php7.4-fpm/oldstable-security,now 7.4.33-1+deb11u5 arm64 [installed,automatic]
php7.4-json/oldstable-security,now 7.4.33-1+deb11u5 arm64 [installed,automatic]
php7.4-opcache/oldstable-security,now 7.4.33-1+deb11u5 arm64 [installed,automatic]
php7.4-readline/oldstable-security,now 7.4.33-1+deb11u5 arm64 [installed,automatic]
php7.4-sqlite3/oldstable-security,now 7.4.33-1+deb11u5 arm64 [installed,automatic]
php7.4-xml/oldstable-security,now 7.4.33-1+deb11u5 arm64 [installed,automatic]
php7.4-zip/oldstable-security,now 7.4.33-1+deb11u5 arm64 [installed,automatic]
php7.4/oldstable-security,oldstable-security,now 7.4.33-1+deb11u5 all [installed,automatic]
php/oldstable,oldstable,now 2:7.4+76 all [installed]

**Your build**
Installation details: Raspberry Pi 4B 4gb
OS: Debian GNU/Linux 11 (bullseye)

svc-user, 2024-04-25:
Hi @tomtom215 

It's quite some time since I ran `apt get upgrade` on my BirdNET-Pi so I just tried it. I too saw a lot of php updates, but BirdNET-Pi continued to work after the upgrade - also after a reboot.

I take you have `shh` access to the Raspberry Pi? If so, what's the output of this two step process?

```bash
cd ~/BirdNET-Pi/scripts && grep -o -E "service_status.*?\..*$" service_controls.php | cut -d'"' -f 2 > services.lst
```
followed by

```bash
while read l; do systemctl status  $l | head -n 3 ;  done < services.lst
```

I have from time to time experienced php failing to startup so for good measure also check the php service;
```bash
systemctl status php7.4-fpm.service
```

tomtom215, 2024-04-25:
Hey @svc-user  thanks for the reply!

Here's the output:

```
● livestream.service - BirdNET-Pi Live Stream
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/livestream.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2024-04-21 19:55:12 CEST; 3 days ago
```
```
● web_terminal.service - BirdNET-Pi Web Terminal
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/web_terminal.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2024-04-21 07:15:30 CEST; 4 days ago
```
```
● birdnet_log.service - BirdNET Analysis Log
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/birdnet_log.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2024-04-21 19:55:10 CEST; 3 days ago
```
```
● extraction.service - BirdNET BirdSound Extraction
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/extraction.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2024-04-21 19:55:10 CEST; 3 days ago
```
```
● birdnet_server.service - BirdNET Analysis Server
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/birdnet_server.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2024-04-21 19:55:11 CEST; 3 days ago
```
```
● birdnet_analysis.service - BirdNET Analysis
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/birdnet_analysis.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2024-04-25 19:43:00 CEST; 877ms ago
```
```
● birdnet_stats.service - BirdNET Stats
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/birdnet_stats.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2024-04-21 19:55:11 CEST; 3 days ago
```
```
● birdnet_recording.service - BirdNET Recording
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/birdnet_recording.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2024-04-25 08:12:33 CEST; 11h ago
```
```
● chart_viewer.service - BirdNET-Pi Chart Viewer Service
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/chart_viewer.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) since Thu 2024-04-25 19:43:06 CEST; 4s ago
```
```
● spectrogram_viewer.service - BirdNET-Pi Spectrogram Viewer
     Loaded: loaded (/home/pi/BirdNET-Pi/templates/spectrogram_viewer.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2024-04-21 19:55:09 CEST; 3 days ago
```

I was surprised it broke anything, as I have always kept my apt packages up to date and usually everything would be fine after a reboot...just not this time

closed 2024-04-25 by tomtom215

svc-user, 2024-04-26:
That's fortunate, @tomtom215 😄 The output is purely diagnostic, the commands makes no changes to anything 🤷‍♂️

I'm glad you're back on track :)