mcguirepr89/BirdNET-Pi #1191: Unsuccessful installation due to unsupported wheel

calumma13, 2024-04-29:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
Installation fails

**Expected behavior**
Successful installation

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
After installation attempt, tried rebooting and accessing web server. Expected bird net page not visible, instead see  Apache2 Debian Default Page. 

Tried rebooting, uninstalling and reinstalling with same error. 

**Your build**
Raspberry Pi 4 with Raspberry Pi OS Lite (64-bit)

**Code or log snippets**
```
Installation fails with following message: 

ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.

Svardsten53, 2024-04-29:
Did you install OS from this file? 

`2023-05-03-raspios-bullseye-arm64-lite.img.zip`

calumma13, 2024-04-29:
I used the OS image directly from the Raspberry Pi imager software. Looks to be a newer release (2024-03-15). I need to use the older file?

psylvyn, 2024-04-30:
I ran into the same issue, and while it is not ideal I was able to get a successful installation using bullseye, rather than the latest bookworm.

calumma13, 2024-05-01:
> I ran into the same issue, and while it is not ideal I was able to get a successful installation using bullseye, rather than the latest bookworm. "Raspberry Pi OS (Legacy, 64 bit) Lite" in the Raspberry Pi Imager

Thanks for that. This also worked for me.

grahamck, 2024-05-01:
I wonder if anyone has a workaround to this other than starting over with a different OS?

georgeh, 2024-05-07:
Installing from `2023-05-03-raspios-bullseye-arm64-lite.img` fixed the issue for me too. 

I chose bookworm originally because https://www.birdweather.com/birdnetpi says [emphasis added]:

> Copy and paste the command below on a Raspberry Pi running the **_latest_** Raspberry Pi OS Lite (64-bit) image to install BirdNET-Pi.

That should probably be updated to be clearer about supported versions.

christopherlock, 2024-05-26:
If you're running RPI5, see my comment here: https://github.com/mcguirepr89/BirdNET-Pi/issues/1194#issuecomment-2132417474

herbbetz, 2024-06-20:
Same error today, when I tried to install the BirdNET-Pi on 'ubuntu-24.04-preinstalled-server-arm64+raspi.img'. I thought to have read somewhere, that BirdNET-Pi prefers ubuntu before bullseye? ChatGPT also thinks, it can be installed with ubuntu...

PaulRB, 2024-07-31:
Same for me. Install fails with Bookworm but succeeds with Bullseye

CP3000-bird, 2025-01-21:
The BirdnetPi by @mcguirepr89 doesn't seem to be updated anymore.
@Nachtzuster forked it and is updating it. I was able to install on Bookworm and it seems to run much smoother and has some more settings to tweak.
I haven't deployed the unit yet in the field so I can't say anything on detection etc.
https://github.com/Nachtzuster/BirdNET-Pi?tab=readme-ov-file