mcguirepr89/BirdNET-Pi #1193: After running successfully for months: Fatal error: Uncaught Error: Call to a member function fetchArray() on bool in /home/admin/BirdNET-Pi/scripts/todays_detections.php:43

mkjn72, 2024-04-30:
After running it very successfully for months I now see this error in the 'Today's detection' page:

`Fatal error: Uncaught Error: Call to a member function fetchArray() on bool in /home/admin/BirdNET-Pi/scripts/todays_detections.php:43 Stack trace: #0 /home/admin/BirdNET-Pi/homepage/views.php(150): include() #1 {main} thrown in /home/admin/BirdNET-Pi/scripts/todays_detections.php on line 43`

Have looked into this a bit. Did a completely new install and tried to migrate files according to this link https://github.com/mcguirepr89/BirdNET-Pi/wiki/Backup-and-Restore-the-Database

I find that the problem seems to be in de birds.db file. As soon as I copy this over the error re-occurs. But I do not have the expertise to invetigate this further. Is there a way to repair this DB file? Or how do I go about trying to resolve the issue. It would really be a shame to lose all historic recordings and information..

Many thanks in advance for any assistance you might offer

christopherlock, 2024-05-26:
Bundled with BirdNET-Pi is a tool to help you administer the sqlite database. Can you see anything here? http://YOUR_PI_HOSTNAME.local/scripts/adminer.php?sqlite=

If so, try running the same SQL query that BirdNET-Pi is running. What happens when you run `SELECT COUNT(*) FROM detections` on the detections table?

Perhaps another thing to try is deleting some of the most recent data that might be corrupt? Hard to say without looking at the database.

hestonhamilton, 2024-06-03:
I am seeing the same error on my home page as of today -

Database is busy

**Fatal error**: Uncaught Error: Call to a member function execute() on bool in /home/admin/BirdNET-Pi/scripts/overview.php:32 Stack trace: #0 /home/admin/BirdNET-Pi/homepage/views.php(424): include() #1 {main} thrown in **/home/demeter/BirdNET-Pi/scripts/overview.php** on line **32**

I tried looking at the original DB but I didn't see anything unusual. I moved that DB temporarily and restarted the server, but the same error is happening even on a fresh/empty DB.

oldgristmill, 2025-03-06:
Anyone figure out what was actually causing this error? I'm dealing with the same error (and others) right now and if its a data corruption error, I'd rather fix the data than delete it, if possible. 

Suggestions?

esparks, 2025-07-29:
Also having the same issue. I'm unable to login anymore either. And the colors of the page have changed.

> Fatal error: Uncaught Error: Call to a member function fetchArray() on bool in /home/pi/BirdNET-Pi/scripts/play.php:420 Stack trace: #0 /home/pi/BirdNET-Pi/homepage/views.php(179): include() #1 {main} thrown in /home/pi/BirdNET-Pi/scripts/play.php on line 420