mcguirepr89/BirdNET-Pi #1194: BIRDNET-PI

gilpayson, 2024-05-01:
BIRBETPI INSTALL FATALLY FLAWED.  DOES NOT INSTALL
A clear and concise description of what the bug is.
Install fails, and the problem is deeply imbedded 
**To Reproduce**
Steps to reproduce the behavior:
1. Try to install BirdNet PI on any machine
2. I have tried 12?  13 images?  
3. read 12 sets of alternate install instrux?
4.Error:
 pip3 install -U -r /home/gil/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.

**Expected behavior**
A clear and concise description of what you expected to happen.
BirdnetPI installed as per the installation instructions
**Screenshots**

**Additional context**
I cannot find, nor have any online docs proved successful.  Somewhere, there is a screwed up modules causing a 32-bit whatzit to nuke the install:

 pip3 install -U -r /home/gil/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.

**Your build**
The hardware on which BirdNET-Pi is running goes here.
Raspberry Pi 5 8GB  64GB SDisk.
**Code or log snippets**
``` pip3 install -U -r /home/gil/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.

```

gilpayson, 2024-05-01:
There are many people with this same problem, and some of them think they found a solution, but actually got randomnly lucky.  I have followed this alternate install instructions, work-arounds, you name it.  Same error every time.

I would like to challenge anyone to follow the install using 64 bit images, do it successfully, and provide dated evidence of an install in May 2024.

matthew73210, 2024-05-02:
This repo isn't being maintained anymore, or so I was under the impression. Have a look at https://github.com/Nachtzuster/BirdNET-Pi

gilpayson, 2024-05-02:
It Worked!  Thank you!

I will now proceed to install two additional systems, and get them ready to
be installed on an uninhabited island in the most remote area of Cape Cod

Jabe, 2024-05-08:
Happened when I tried Bookworm yesterday, use Bullseye instead, which is however not supported on the Pi 5.

christopherlock, 2024-05-26:
I spent a whole day fixing this issue on my RPI5, and eventually succeeded. I'll try to set out what I did below. However, **if you're experiencing this issue on your RPI5, try the fork https://github.com/Nachtzuster/BirdNET-Pi as these issues appear to have been resolved**. I wish I'd have found this sooner as it would have saved me a whole lof of time! On the other hand, it was fun finding my way to the solution.

Now, onto the fix for my RPI5.

The problem is that BirdNET-Pi makes some incorrect assumptions about the versions of software installed on the RPI5, such as python and php. The RPI5 comes installed with Python 3.11, but this version of BirdNET-Pi requires Python 3.9 to run.

Fix the python installation
===========================
First, download and instal Python 3.9:

	 # install prerequsite packages
	 sudo apt-get install  libncursesw5-dev libssl-dev libsqlite3-dev tk-dev libgdbm-dev libc6-dev libbz2-dev  libffi-dev liblzma-dev

	 # download, compile and install python 3.9
	 wget https://www.python.org/ftp/python/3.9.19/Python-3.9.19.tgz
	 gunzip Python-3.9.19.tgz 
	 tar xf Python-3.9.19.tar 
	 cd Python-3.9.19/
	 ./configure
	 sudo make altinstall 	# altinstall ensures your existing system-wide python remains untouched

	 # you should now have Python 3.9 installed
	 /usr/local/bin/python3.9 --version

Next, repeat some of the BirdNET-Pi installation steps (manually)

	 cd $HOME/BirdNET-Pi/
	 # remove the original virtual env built with the incorrect version of Python and re-create it
	 rm -Rf birdnet
	 /usr/local/bin/python3.9 -m venv birdnet

	 # activate the virtual env
	 source ./birdnet/bin/activate

	 # Install the pip modules
	 pip3 install -U -r $HOME/BirdNET-Pi/requirements.txt

There should no longer be errors in the BirdNET-Pi log file and, if your mic is hooked up, the logs should even show BirdNET-Pi properly analyzing bird sounds. Once the UI issue is resolved (see below), the log file is available to view at http://YOUR_PI_HOSTNAME.local/log/

Fix the Caddy UI
================
The UI http://YOUR_PI_HOSTNAME.local is initially broken on the RPI5 for several reasons.

Firstly, Caddy (the web server) fails to start because Apache is already running and hogging port 80. You can check this by running `sudo systemctl` to see failed the failed caddy service and the running apache2 service.

	# Stop and disable apache2
	sudo systemctl stop apache2
	sudo systemctl disable apache2

Next, the version of php on the RPI5 is newer than Caddy expects 

	# Check which version of php-fpm is installed
    ls /run/php
	php8.2-fpm.pid  php8.2-fpm.sock  php-fpm.sock

	# edit the Caddy config file to point to this version
	sudo vi /etc/caddy/Caddyfile

	#
	# old
	#  php_fastcgi unix//run/php/php7.4-fpm.sock
	#
	# new
  	#  php_fastcgi unix//run/php/php8.2-fpm.sock
	#

Finally, the user that the caddy service and php run as ('caddy') doesn't have access to read the necessary BirdNET-Pi files, despite the fact that the user 'caddy' is part of the 'pi' group. I don't really know how this ever worked. A quick fix is to grant the entire pi directory read and execute access for other users in the pi group.

	# Confirm the caddy user is unable to read the index file
	sudo -u caddy ls -lrta /home/pi/BirdSongs/Extracted/index.php

	# fix the permissions
	sudo chmod -R g+rx /home/pi

	# it should work now
	sudo -u caddy ls -lrta /home/pi/BirdSongs/Extracted/index.php

	# now restart the failed caddy service
	sudo systemctl start caddy

Head over to http://YOUR_PI_HOSTNAME.local/ and things should now be working! Perhaps try a reboot (`sudo reboot`) if not.

Other notes:
 - Check out the BirdNET-Pi log at http://YOUR_PI_HOSTNAME.local/log/
 - I couldn't find where caddy was logging to, so I added a "log" section to the conf file. This helped me debug.

```
http://  {
  root * /home/pi/BirdSongs/Extracted
  file_server browse
  handle /By_Date/* {
    file_server browse
  }
  handle /Charts/* {
    file_server browse
  }
  reverse_proxy /stream localhost:8000
#  php_fastcgi unix//run/php/php7.4-fpm.sock
  php_fastcgi unix//run/php/php8.2-fpm.sock
  reverse_proxy /log* localhost:8080
  reverse_proxy /stats* localhost:8501
  reverse_proxy /terminal* localhost:8888

  log {
    output file /var/log/caddy/mine.log {
      roll_size 1gb
      roll_keep 5
      roll_keep_for 24h
  }
}
```

technowannabe, 2024-12-28:
I had the same issue and was able to fix it just be shutting down/disabling Apache and starting up Caddy. Thanks!

0xJustin, 2025-01-28:
Ditto! Shutting down Apache and restarting Caddy was the solution for me!

joaldes, 2025-03-03:
Thanks @christopherlock!  These instructions got me going again!