mcguirepr89/BirdNET-Pi #1198: Apache2 Debian Default Page

R3tr0BoiDX, 2024-05-09:
**Describe the bug**
After installation, only the Apache2 Debian Default Page shows up

**To Reproduce**
Steps to reproduce the behavior:
Follow the Pi Zero 2 install [guide](https://github.com/mcguirepr89/BirdNET-Pi/wiki/RPi0W2-Installation-Guide)

**Expected behavior**
The BirdNET-Pi default page should load...ig?

**Screenshots**
[image]

**Additional context**
I've followed the installations process exactly as described. I flashed brand new RPi OS Lite x64 to the SD card, booted it, entered the command to increase the swap file size and to disable the Wi-Fi power safe (as given in the Pi Zero 2 setup), rebooted and then progressed with the normal install, using the bash scrip:
```
curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash
```
The terminal prompt is adjusted when I log in via SSH, it just seems like the webpage is not booting up.

**My build**
Raspberry Pi Zero 2 with RPi OS Lite x64 on a 32 GB SD card

fowlerjk, 2024-05-11:
Same issue, and I followed the instructions [here](https://github.com/Nachtzuster/BirdNET-Pi?tab=readme-ov-file) when the original installation did not work.

Juliedu45, 2024-05-14:
idem ici comment faire pour résoudre le soucis ?

cube4you, 2024-05-15:
Same here on RPI 4

alexbelgium, 2024-05-16:
Hi, use this fork which is the new active one : https://github.com/Nachtzuster/BirdNET-Pi

fowlerjk, 2024-05-19:
> Hi, use this fork which is the new active one : https://github.com/Nachtzuster/BirdNET-Pi

As noted in my post above, I tried this fork already.

christopherlock, 2024-05-26:
The likely cause of this is that Apache is hogging port 80 and preventing the Caddy web server is unable to start. You need to stop apache and restart Caddy.  See my comment here: https://github.com/mcguirepr89/BirdNET-Pi/issues/1194#issuecomment-2132417474