mcguirepr89/BirdNET-Pi #1199: Question: how to get plain text in a https/jsons webhook?

jeroenvano, 2024-05-13:
Hi all, I've been playing around to get some apprise notifications pushed to a webhook but can't get it to work as plain text what it needs.

Originally it is a http/https receiving webhook:
https://user:pass@host.domain.com:443/local/path/to/infoticker.cgi?service_id=16&text=
Using this in the BirdNET-Pi apprise config as http or https I get:
2024-05-13 12:13:01,384 - ERROR - Unparseable URL https://{URL}

Using json
https://user:pass@host.domain.com:443/local/path/to/infoticker.cgi?service_id=16&-text=
It passes and works but I get the full json output as plain text
{"version": "1.0", "title": "Message Title", "message": "Message Body", "attachments": [], "type": "info"}

How can I get only the body (message) text as an output on the webhook? 
If someone can point or direct me into the right direction that would be highly appreciated!

Cheers,
Jeroen

audevuilli, 2024-05-13:
Hi @jeroenvano 

Not sure how you try to do this, using javascript, python? 

But this link might be helpful: [https://mkyong.com/javascript/how-to-access-json-object-in-javascript/](https://mkyong.com/javascript/how-to-access-json-object-in-javascript/ )

jeroenvano, 2024-05-13:
> Hi @jeroenvano
> 
> Not sure how you try to do this, using javascript, python?
> 
> But these links might be helpful:
> 
> 1. https://mkyong.com/javascript/how-to-access-json-object-in-javascript/
> 2. https://www.w3schools.com/js/js_json_objects.asp

@audevuilli I am usnig a standard BirdNET-Pi installation where Apprise Notifications can be setup and enabled under Tools-> Settings. Each service should be on its own line, so you just paste the lines and save.
e.g. 
"[https://user:pass@host.domain.com:443/local/path/to/infoticker.cgi?service_id=16&text=](https://user:pass@host.domain.com/local/path/to/infoticker.cgi?service_id=16&text=)"