mcguirepr89/BirdNET-Pi #1202: Update Wiki to specify the need to install Bullseye and not Bookworm

noahtolsen, 2024-05-18:
The current version of Raspberry PI OS LIte (64 bit) causes an error with the install script. Updating the wiki to specify to use the legacy Bookworm would avoid new users using the wrong version of Raspberry Pi OS.

alexbelgium, 2024-05-18:
Hi, I am voir affiliated with this fork but would highly recommend it : https://github.com/Nachtzuster/BirdNET-Pi

It solves and improves many things including bookworm support

alexbelgium, 2024-05-18:
Hi, I would highly recommend to try this fork : https://github.com/Nachtzuster/BirdNET-Pi

It solves many issues including bookworm support

https://github.com/mcguirepr89/BirdNET-Pi/discussions/1177#discussioncomment-9129759

noahtolsen, 2024-05-18:
I had tried the fork when I first ran into trouble with this repo. It seemed to work ok but I was seeing a lot of discrepancies between what the log was saying it was hearing vs the final predictions that were outputted and written to the db. But I was also tinkering with my mic a lot at the time. Maybe I'll give it another go.

alexbelgium, 2024-05-18:
I discovered Birdnet-pi quite recently, but find this app developed by mcguirepr89 so incredible that I have now devoted a lot of time to understand it a bit better.

As I understand, the main differences that could affect predictions would be the optional implementation of the model 2.4, and especially the 2.4v2 that relies on a probability of detection according to eBirds checklists. Honestly based on my experience is it quite good for very common birds, but has quite bad biases against bird of preys and night birds. To avoid that either you could use the model BirdNET_6K_GLOBAL_MODEL (same as this version of B-Pi), the BirdNET_GLOBAL_6K_V2.4 without enabling the Species range model (enabled by default) or use a very low threshold (I'm using 0,0007).

Apart from that, indeed I see also many quite erroneous elements in the logs but I think it is just to show that the model is working, there is also the confidence of prediction shown next to it that should be quite low for such species.

I have also "trained" a bit my system by modying the exclusion list to compensate for species that I know for sure are misdetections and would not occur in this area (after listening to the bird songs).

All that said, the developer is quite responsive ! Which basically is the key issue here that although mcguirepr89 has developed a near perfect system he doesn't seem to hang around anymore to support the systems...

Well in the end I guess it's whatever works for everyone ; there is not one ultimate solution for all ;-) have a nice day