mcguirepr89/BirdNET-Pi #1249: Mono or Stereo

SchmidL, 2024-09-16:
Hi are there any insights of using a mono or a stereo microphone setup with the BirdNET-Pi?
What would be the pros for a stereo, so far i could not figure out any, besides more data :)

tphakala, 2024-09-16:
BirdNET Analyzer works with mono files only, if anything stereo could introduce phase issues if stereo channels are downmixed to mono. So mono is recommended.

Btw this repository is abandoned, further development is done on https://github.com/Nachtzuster/BirdNET-P
there is also my golang based implementation https://github.com/tphakala/birdnet-go

SchmidL, 2024-09-16:
Thank you a lot! just saw many stereo setups, but yeah, was not so conclusive for me :)

closed 2024-09-16 by SchmidL