mcguirepr89/BirdNET-Pi #1255: New install stops working

bigfish69, 2024-10-11:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
Have recently installed BirdNet Pi and it stops working after a few minutes to a couple of hours.  Have reinstalled a few times.  Have even installed on multiple microSD cards

**To Reproduce**
Steps to reproduce the behavior:
1. Open birdnetpi.local
2. View Log
3. Comes up with ...

Have reinstalleed BirdNET Pi a few times, even on different microSD cards but get the following only after a few minutes to an hour

18:56:46--- sig, rate = librosa.load(path, sr=sample_rate, mono=True, res_type='kaiser_fast')
18:56:46--- File "BirdNET-Pi/birdnet/lib/python3.9/site-packages/librosa/util/decorators.py", line 88, in inner_f
18:56:46--- return f(*args, **kwargs)
18:56:46--- File "BirdNET-Pi/birdnet/lib/python3.9/site-packages/librosa/core/audio.py", line 170, in load
18:56:46--- y, sr_native = __audioread_load(path, offset, duration, dtype)
18:56:46--- File "BirdNET-Pi/birdnet/lib/python3.9/site-packages/librosa/core/audio.py", line 226, in __audioread_load
18:56:46--- reader = audioread.audio_open(path)
18:56:46--- File "BirdNET-Pi/birdnet/lib/python3.9/site-packages/audioread/init.py", line 132, in audio_open
18:56:46--- raise NoBackendError()
18:56:46---audioread.exceptions.NoBackendError

Any idea of how to resolve this please?

**Expected behavior**
See recording, analysing etc

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem or your installation here.

**Your build**
Raspberry Pi4, model B
Using the following OS
[image]

**Code or log snippets**
```
Paste your 
log output 
or 
code here
```

bigfish69, 2024-10-11:
Tried to reinstall again but this time the microsSD card memory was not only reduced from 64GB to 255MB but is no longer able to be formatted

alexbelgium, 2024-12-18:
You should use this fork, much more up to date : https://github.com/Nachtzuster/BirdNET-Pi/ or this other version : https://github.com/tphakala/birdnet-go