mcguirepr89/BirdNET-Pi #1256: Species and Detection Stats not showing up on BirdWeather

Techbirder, 2024-11-02:
Everything is running smoothly on my RPi3b locally and I have not issues there. My station is publicly registered on BirdWeather, and on the BirdWeather map the species and detection stats always display 0 for my station, despite showing a “Last Detection” accurately. I have tried restarting services, but everything seems fine locally. It seems there is a disconnect when the data is pulled to be posted on BirdWeather. 

FYI…I appreciate all your work and info on this. It’s great fun!

[image]

Svardsten53, 2024-11-02:
Verify you have defined the same latitude and longitude information in BirdWeather as you have in BirdNET-Pi settings.

Techbirder, 2024-11-03:
I have verified that my lat and long are identical in both BirdWeather and BirdNET-Pi. I also made sure that they were both set to 4 decimal places as noted in BirdNET-Pi.

Techbirder, 2024-12-04:
Any other ideas on this? Instead of 4 decimal places, I have tried 3 with the same result. I noticed another RPi3B geographically close to mine that seems to have the same issue. It’s strange that the time for the last detected species displays correctly. It’s just the “species” and “detections” that always display 0.

static418, 2025-03-27:
Having the same issue.  No errors or info in the logs, just 201s to soundscape but nothing ever seems to upload.  Verified GPS coordinates are the same, and also experiencing birdweather showing my last detection as moments ago but my 24 hour detections as 0 and no birds.

nobbyneb, 2025-04-26:
I'm having the same issue also. @Techbirder and @static418 were you able to resolve? Have checked the coordinates match. I am getting a 'last detection' but no species listed.

What were your confidence levels, may I ask? Any advice appreciated, thank you.

mattisj1, 2025-04-26:
I have just the same problem. Coordinates checked. My station is #13260.

On the right lower corner it does not notice my station, though I can see the green pointer on the screen.

It is the same with Piper's station.

matti

pfbangs, 2025-04-26:
Same @nobbyneb @mattisj1 (what are your settings? are your stations new?)-- green icon exists visually on map (but not included in station count bottom right), shows accurate last detection time, but nothing else @ species/detection count. Station #13231
GPS accurate across the two platforms, BirdWeather station Token is being used as "BirdWeather ID" in BirdNET Pi settings. 
Overlap: 0
Minimum Confidence: .7
Sigmoid Sens: 1.25
(human) Privacy threshold: 1%

Station/connection has been up for over 24h now (it is new). So unless there's some new or general data update interval (to birdweather) which is larger than 24h, I'm stumped. Will look into checking logs directly if needed

nobbyneb, 2025-04-27:
Thanks @pfbangs 

Yes, i'm new this week. #13223 

GPS Coordinates and decimal places cross-checked. Screenshot below.

[image]
[image]
[image]
[image]
[image]
[image]

Techbirder, 2025-04-27:
I just checked and my system (#7973/Piper) is still experiencing the same
problem as when I first posted. My settings are all similar to what
everyone else is posting. I’m running a Pi3b, so not sure if that has any
bearing.

My system has been running for several months now with confidence ranging
from 70-90%. Although I have a great horned owl nesting in close proximity
(40’) and I have yet to record a call. I also had a Towhee calling very
loudly this afternoon and that one wasn’t captured either. That may be the
nature of the directionality of my microphone and not this topic.

It would be great to get to the bottom of this. It’s the difference from
being individual or part of something bigger.

Perry Yingling
***@***.***
(717)422-6441

mattisj1, 2025-04-28:
Hi,

my station had started last night. Now it shows  as a station al low left corner. Also species loaded.

I did not do anything, neither touch on blueberry, for I was away 36 hours.

No clue what happened. But when something works don't touch on anything.

Good luck with Yours.

Cheers: matti

nobbyneb, 2025-04-28:
Congrats @mattisj1 , yes your station looks good now. 

How old is your station and birdweather account may i ask? Would you mind sharing your settings, similar to mine above? Particularly interested in coordinates decimal places, as well as your selected model, privacy threshold and Birdnet Lite settings. thank you.

Many thanks

tphakala, 2025-04-28:
Birdweather ignores all coordinate settings from BirdNET-Pi submissions, so it does not matter what you have set. I would also like to point out that this repository is abandoned and will eventually stop working when Birdweather stops accepting wav file uploads, as they are moving to flac only.

nobbyneb, 2025-04-28:
> Birdweather ignores all coordinate settings from BirdNET-Pi submissions, so it does not matter what you have set. I would also like to point out that this repository is abandoned and will eventually stop working when Birdweather stops accepting wav file uploads, as they are moving to flac only.

Very helpful thank you. Does this essentially mean Birdweather will reject integration from Birdnet-PI completely, at some stage?

tphakala, 2025-04-28:
Unless BirdNET-Pi code is updated to match new requirements of BirdWeather API then yes, BirdWeather integration will stop working. It is unlikely that this repository will get any updates, but https://github.com/Nachtzuster/BirdNET-Pi may and likely will. I also have implemented this change on my BirdNET-Go application https://github.com/tphakala/birdnet-go.

mattisj1, 2025-04-28:
Hi @nobbyneb,

my station has been running from the 25.4.2025. As birdweather id I have copied the token-code from the BirdWeather. Below settings You asked.

Good Luck.

-matti-

[image]

pfbangs, 2025-04-28:
> Unless BirdNET-Pi code is updated to match new requirements of BirdWeather API then yes, BirdWeather integration will stop working. It is unlikely that this repository will get any updates, but [...]

For visibility, yesterday, I modified reporting.py to double all (Birdweather reporting) timeout intervals (this did not resolve the issue).

Originally, the Birdnet-Pi logs showed:

`[utils.reporting][ERROR] Cannot POST soundscape: HTTPSConnectionPool(host='[app.birdweather.com](http://app.birdweather.com/)', port=443): Read timed out. (read timeout=30)`
 
After doubling the timeout intervals, it then showed:
```
13:10:36---[utils.reporting][INFO] Soundscape POST Response Status - 504
13:10:36---[utils.reporting][ERROR] Server returned error code: 504
```

...suggesting the requests are simply timing out completely on the Birdweather side/remote server. 

+++++++++++++++

I reached out to BW ([via the email address mentioned in this thread](https://github.com/mcguirepr89/BirdNET-Pi/discussions/1147#discussion-6323096)) and at Tim's request, I provided POST (JSON) captures of some very high-confidence detections and soundscape POSTs. This was done by further modifying the reporting.py file to capture much more verbose detail around the POST traffic in the Birdnet-Pi log.

```
def bird_weather(file: ParseFileName, detections: [Detection]):
    conf = get_settings()
    if conf['BIRDWEATHER_ID'] == "":
        return
    
    # Set up logging for birdweather POSTs
    import logging
    import json
    bw_logger = logging.getLogger('birdweather_posts')
    bw_logger.setLevel(logging.DEBUG)
    # Create file handler if it doesn't exist
    if not bw_logger.handlers:
        file_handler = logging.FileHandler('/home/rich/birdweather_posts.log')
        formatter = logging.Formatter('%(asctime)s - %(levelname)s - %(message)s')
        file_handler.setFormatter(formatter)
        bw_logger.addHandler(file_handler)
    
    if detections:
        # POST soundscape to server
        soundscape_url = (f'https://app.birdweather.com/api/v1/stations/'
                          f'{conf["BIRDWEATHER_ID"]}/soundscapes?timestamp={file.iso8601}')
        with open(file.file_name, 'rb') as f:
            wav_data = f.read()
        gzip_wav_data = gzip.compress(wav_data)
        
        # Log the soundscape POST request (can't log binary data easily, so just log the URL)
        bw_logger.debug(f"SOUNDSCAPE POST to {soundscape_url}")
        bw_logger.debug(f"Headers: Content-Type: application/octet-stream, Content-Encoding: gzip")
        bw_logger.debug(f"Binary audio data length: {len(wav_data)} bytes, Compressed: {len(gzip_wav_data)} bytes")
        
        try:
            response = requests.post(url=soundscape_url, data=gzip_wav_data, timeout=60,
                                     headers={'Content-Type': 'application/octet-stream', 'Content-Encoding': 'gzip'})
            log.info("Soundscape POST Response Status - %d", response.status_code)
            
            # Log the response
            bw_logger.debug(f"Soundscape POST Response Status: {response.status_code}")
            bw_logger.debug(f"Soundscape POST Response Headers: {json.dumps(dict(response.headers), indent=2)}")
            
            # Check if response status is not successful
            if response.status_code >= 400:
                log.error("Server returned error code: %d", response.status_code)
                bw_logger.error(f"Server returned error code: {response.status_code}")
                return
                
            # Check if response has content before parsing JSON
            if not response.content:
                log.error("Empty response received from server")
                bw_logger.error("Empty response received from server")
[... 50 lines cut ...]
                        bw_logger.debug(f"Detection POST Response: {json.dumps(resp_data, indent=2)}")
                    except:
                        bw_logger.debug(f"Detection POST Response (not JSON): {response.text[:500]}")
                
            except BaseException as e:
                log.error("Cannot POST detection: %s", e)
                bw_logger.error(f"Cannot POST detection: {e}")
```

I haven't heard back after providing those logs to BW. 

+++++++++++++++

I blew away my Birdnet-Pi instance this morning, and stood up a Birdnet-Go instance in its place. I am now using FLAC on GO (instead of WAV on Birdnet-Pi). When using my same/original BW station (token/ID), BW exhibited the same behavior/partial data on the station. The Birdnet-Go "test connection" utility for the BW integration shows all success:

[image]

I moved (GPS) my original station a small amount and created a new station on BW (perhaps I screwed something up when setting up the original station, or perhaps there was some provisioning issue on the BW backend with my original station). Assigning my new station token to Birdnet-Go also shows all success @ the BirdWeather integration "Test Connection" checks. The new BW station, however, shows the exact same behavior (not populating detection counts/detail, or including the station in the global station count/index in the BW UI, and the "Last Detection" time is also not updating in a timely/accurate manner in response to high-confidence detections). 

[image]

I'm simply putting this information here to communicate it appears this issue (BW stations not fully populating detection data) does not appear (to me) to be specific to the Pi-based platform (Birdnet-Pi, Birdnet-Go, etc). 

++++++++
If anyone has any ideas or can point me to a more relevant discussion space for this, please let me know & ty

Techbirder, 2025-04-29:
Wow! Very thorough. Thanks for all the work! A little beyond my
capabilities. I’m grateful (and a little envious) of people like you.
Hopefully you get a response from BW. I was unaware of Birdnet-Go and FLAC.
You got me searching now.

Perry Yingling
***@***.***
(717)422-6441

spohlenz, 2025-04-29:
I've been investigating some of these issues on the BirdWeather side of things and I have hopefully found the cause. A recent change last week meant that BirdNET probabilities weren't being computed on new BirdNET-Pi stations. That should now be fixed and those stations with missing probabilities manually updated. This will only impact new detections though.

@pfbangs The 504 timeouts you have been seeing are a different issue though. The soundscape updates do appear to be completing on the server (eventually) but obviously you are not getting the required response needed to then associate the detections. Do you know if your Pi setup has a strong Wifi signal and/or decent upload speed? Given that the smaller FLAC uploads are completing okay, that is my best guess as to what was happening with the WAV uploads.

pfbangs, 2025-04-29:
@spohlenz Thank you **so** much, Sam. Do I have the ability to DEL my old/original station (13231)? Simply don't want to clutter the map, if it's within my control at all (if not, perhaps it is a feature request?). 

Speeds from the Pi, averaged across 3 tests (range was very narrow)
DOWN: 15.58 Mbps | 1.9475 MB/s
UP: 14.47 Mbps | 1.80875 MB/s

It (the Pi) is certainly a distance from my WAP and is on the 2.4ghz band/SSID for distance (5ghz doesn't reach), so it absolutely could be that (although I don't recall the WAV upload sizes and would have expected those to complete within the timeout windows, especially after doubling them). Do you recall how big the soundscape/detection (WAV) uploads might typically be (just for community awareness/visibility)? It may be worth adding a visible disclaimer and quick "install/run speedtest" guide to confirm-- somewhere/anywhere to indicate some recommended minimum bandwidth as a potential bottleneck/point of failure if not already present (in the station creation/management UI or some FAQ, perhaps). 

For others' visibility: 

Install Speedtest on bookworm:
`sudo apt install speedtest-cli`

Run Speedtest:
`speedtest-cli --secure`

It looks like I'm all set. The BW station index is now including my (new) station in the count and detections are being passed and updated pretty quickly. Very happy to be able to share the station externally now! Thank you again, Sam-- really appreciate it!

[image]

e-lbarto, 2025-04-29:
I had the same problem:
The station was displayed correctly on the map, including "Last detection," but the number of species detected remained at "0." I meticulously followed all the tips I read (shorten the station coordinates entered in Birdnet-Pi and birdweather.com to 4 decimal places), and enter the Birdweather token (from birdweather.com) in Birdnet-Pi as "Birdweather ID" (quite confusing).

I deduced the following without reading any details about it—but it ultimately led to a solution to the problem. The cause is additional (absolutely sensible) protection of the Birdweather network against being flooded with nonsensical detections.
Birdweather calculates a "score" from "confidence" (for the Pi's correct detection of the species) and "probability" (for the probability that the bird species even occurs at these coordinates). Only when a certain score is exceeded does birdweather transfer the station's data to the network.
In my case, the confidence was initially set too low; in addition, the station frequently registered the calls of the collared dove from the neighboring property. The collared dove has a probability of only 35% at the station location in birdweather – but Birdnet doesn't know that they live in the neighbor's aviary. Increasing the confidence (ideally between 0.7 and 0.8) and reorienting the microphone solved the problem – after a few hours, the station was accessible in the birdweather network. I hope this helps! elbarto (elbirdo)

e-lbarto, 2025-04-29:
Just read: My conclusions coincide with Spohlenz's findings. So if you raise the confidence in your Birdnet-Pi until there is an update for the system

e-lbarto, 2025-04-29:
@spohlenz ooh, now I understand ! just noticed that you fixed the problem by updating the pi-stations just a few hours ago - also mine on my terrace. so this is why it was working suddenly this morning. So thank you very much, it´s such a good project, I´m going to set up 2 birdnet-Pis in our school gardens (Erlangen, Bavaria, Germany)

nobbyneb, 2025-05-04:
Thanks so much for your help with this! @spohlenz