mcguirepr89/BirdNET-Pi #1278: Specify that one should use the legacy Bullseye version RPi OS rather than Bookworm

alanhoyle, 2025-02-04:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
Raspberry Pi OS 12 "Bookworm" is not compatible with BirdNET-Pi.  The installation instructions should specify that we need to start with the Legacy Bullseye version.  See #847 

Step 3 should specify that we should use the Legacy 64bit 

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Follow instructions to install BirdNET-Pi in the wiki: https://github.com/mcguirepr89/BirdNET-Pi/wiki/Installation-Guide
2. On step 19 the failure occurs.

**Expected behavior**

BirdNET-Pi installs.

**Your build**
RPI Zero 2W

**Code or log snippets**
```
++ VIRTUAL_ENV_PROMPT='(birdnet) '
++ export VIRTUAL_ENV_PROMPT
++ '[' -n /usr/bin/bash -o -n '' ']'
++ hash -r
+ pip3 install -U -r /home/birder/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.
birder@birdnetpi:~ $ uname -a
Linux birdnetpi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
```

compare with the Legacy OS:

```
birder@birdnetpi:~ $ uname -a
Linux birdnetpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
birder@birdnetpi:~ $ curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash
[... snip ...]
++ export PS1
++ '[' -n /usr/bin/bash -o -n '' ']'
++ hash -r
+ pip3 install -U -r /home/birder/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl
Requirement already satisfied: pip in ./birdnet/lib/python3.9/site-packages (from -r /home/birder/BirdNET-Pi/requirements.txt (line 1)) (20.3.4)
Collecting pip
[ .... ]
Archive:  /home/birder/BirdNET-Pi/model/labels_nm.zip
  inflating: /home/birder/BirdNET-Pi/model/labels_en.txt
+ exit 0
Installation completed successfully
```

Fitzroyr, 2025-02-06:
BirdNET-Pi seems to only run with Python 3.9. Bullseye comes with this version, Bookworm i believe is python 3.11 or 3.12.  That said, I found that Bullseye lite has an issue connecting to WiFi , if you are running a headless system you will need to connect a monitor and keyboard to configure WiFi. I got around this by using Bullseye Full.

alanhoyle, 2025-02-06:
If I used the legacy Bullseye, everything installed and it technically worked on my RPi Zero2W.  No need to plug in a monitor or install the full version:  Both installation attempts were over WiFi using SSH.  

Sidenote:  It wasn't performant on the hardware (too slow:  the web interface took like 30 seconds to respond, if it responded at all), but it wouldn't even install with the non-legacy Bookworm OS.  

It works much better with an RPi 4:   I think the Zero2 was thrashing to the card.

alexbelgium, 2025-02-12:
Hi, just use https://github.com/Nachtzuster/BirdNET-Pi it works with bookworm and is much more up to date

alanhoyle, 2025-02-14:
I am switching to the updated fork from @Nachtzuster.  Thanks @alexbelgium

closed 2025-02-14 by alanhoyle