mcguirepr89/BirdNET-Pi #1293: Can't access Tools

Jon-Roads, 2025-04-26:
I've set up the Birdnet Pi using Nachtzuster, and it's working and I can access the interface from my phone and laptop. But when I try and get into the Tools section is asks for my username and password, and when I put them in it goes straight back to the username and password screen. It'll endlessly do that until I click Cancel, and then just say "You cannot edit the settings for this installation". Has anyone else had this, and how do I fix it?

Svardsten53, 2025-04-27:
I also have this problem since I switched to Nachtzuster. The solution is to enter birdnet and password several times, sometimes just birdnet.
Since it concerns Nachtzuster, you should describe the error there (https://github.com/Nachtzuster/BirdNET-Pi/issues) and refer to this record on McGuire.

Jon-Roads, 2025-04-27:
Ah thanks for that, I shall do that. Entering it several times definitely doesn't work though, it's just an infinite loop where I enter the username and password, click submit, and it immediately flashes back to the enter the username and password screen. But I'll put a post up in the correct forum. Thanks!